luxonis/depthai-ros #568: [custom] modifying stereo parameter with dynamic reconfig

khuechuong, 2024-07-29:
I'm trying to figure out where I can do this:

[image]

which launch file does that? I'm using ROS Noetic

Serafadam, 2024-07-30:
Hi, this is done via [rqt tool](https://wiki.ros.org/rqt), more specifically dynamic_reconfigure plugin

khuechuong, 2024-07-30:
@Serafadam yes, but when I ran the camera.launch and did rqt, and add it using image view and dynamic_reconfigure plugins, I didn't see this oak.

khuechuong, 2024-07-30:
@Serafadam 
[image]

khuechuong, 2024-08-05:
@Serafadam any updates?

Serafadam, 2024-08-06:
Hi, sorry, I had written a response but it seems it wasn't published. To get the full list of parameters you need to click on `oak`, ROS1 parameters aren't namespaced properly as in ROS2 due to some limitations.

khuechuong, 2024-08-06:
@Serafadam as u can see in this picture [image], 

that's all the parameter that's available to me in dynamic reconfigure. I don't see the depth or rgb or stereo parameters like confidence or IR, etc

khuechuong, 2024-08-06:
nvmd got it

closed 2024-08-06 by khuechuong