luxonis/depthai-ros #570: OAK-D PRO don't have depth topic

Ragnar7982, 2024-08-05:
Hello, I'm using Turtlebot4 and OAK-D PRO, OS is Ubuntu22.04 and ROS2-humble, when I call ros2 topic list, the camera only have these topic:
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora

Is there any NOT compressedDepth topic? Because I wanna use RTAB-map, so is there any settings can change to depth(not compressed)? Thank you.

Serafadam, 2024-08-05:
Hi, this can be the result of your config file, more specifically pipeline type. You can find more on pipeline types [here](https://docs.luxonis.com/software/ros/depthai-ros/driver/#Pipeline%20parameters)

Ragnar7982, 2024-08-05:
@Serafadam  Hello, thanks for replying fast. When I run camera.launch.py, it shows:
[component_container-1] [ERROR] [1722860415.251114816] [oak]: No available devices
[component_container-1] [INFO] [1722860415.251352386] [oak]: No ip/mxid specified, connecting to the next available device.
Is there any way can fix this?

Serafadam, 2024-08-05:
Hi, I think it would be best if you go through the steps described in our [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md). You can also check [this](https://docs.luxonis.com/hardware/platform/deploy/usb-deployment-guide/#Debugging) page for general information regarding connectivity issues

Ragnar7982, 2024-08-07:
Hello, sorry for the late reply, so after I connect to laptop the camera works find(USB3.0),but why it's wrong with rtabmap, all of settings are default, and I run: ros2 launch depthai_ros_driver rtabmap.launch .py, but it feels so lagging, and it shows very weird, do you know where might be the problem?
[image]

Serafadam, 2024-08-07:
Hi, based on this image I think it is related to odometry losing track of features, this can be cause by fast movements or exposure changes, you can retrigger the odometry via ROS service or RTABMap's GUI. Regarding latency, it is probably related to ROS2 DDS, you can find some more information on improving the latency [here](https://docs.ros.org/en/jazzy/How-To-Guides/DDS-tuning.html)

Ragnar7982, 2024-08-07:
Thanks for replying fast, but I run visualized_pointcloud.py( default setting), which is in depthai_ros_driver, it shows like this, I am wondering if it's camera problems?
This is far:
[image]
This is close:
[image]
These are actually similar to rtabmap

Ragnar7982, 2024-08-07:
@Serafadam Did you ever try rtabmap before? If you did it do you mind let me see your parameters? Thank you.