luxonis/depthai-ros #571: lower the max range

khuechuong, 2024-08-06:
Is there a way to set the maximum range? Let's say I just want it to see up to 10 meters. Can I set a param?

Serafadam, 2024-08-06:
Hi, in `depthai_ros_driver` you can do that using `threshold_filter`:
```
i_enable_threshold_filter: true
i_threshold_filter_min_range: 400
i_threshold_filter_max_range: 15000
```
I see that [documentation](https://docs.luxonis.com/software/ros/depthai-ros/driver/#DepthAI%20ROS%20Driver-List%20of%20parameters) needs to be updated with those parameter names/values, but you can also refer to [param_handler](https://github.com/luxonis/depthai-ros/blob/82ce67cf91182b11b26504af3973dc312824970e/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp#L152) classes for the values available

khuechuong, 2024-08-06:
@Serafadam what unit is it in? mm?

khuechuong, 2024-08-06:
@Serafadam also declaring 
```
i_enable_threshold_filter: true
i_threshold_filter_min_range: 400
i_threshold_filter_max_range: 15000
```
in ```camera.yaml``` doesn't mean anything since that code only sets i_threshold_filter_min_range and max range when i_enable_threshold_filter is false.

Serafadam, 2024-08-07:
Hi, yes regarding mm, and regarding filters, the default value is false but it can be overridden with a parameter. Additionally, I might've been too unspecific, since this parameter is related to stereo, you have to append `stereo` name tho the parameters, as in other nodes, so:
```yaml
stereo_i_enable_threshold_filter: true
stereo_i_threshold_filter_min_range: 400
stereo_i_threshold_filter_max_range: 15000
```

khuechuong, 2024-08-07:
ya I did set it under stereo. but nothing changes, the pointcloud still shows points beyond the max_range. I'm using the noetic branch by the way

Serafadam, 2024-08-08:
Just to be sure, which launch file are you using?

khuechuong, 2024-08-08:
rgbd_pcl.launch and I modify the rgbd.yaml

Serafadam, 2024-08-13:
Hi, I was unable to replicate the issue, could you show some logs by running `export DEPTHAI_DEBUG=1`? This should output whether those parameters are changed

khuechuong, 2024-08-21:
@Serafadam I'm having that issue again. I set it 
```
/oak_d_pro_poe_front:
    camera_i_mx_id: 1844301041E8E9F400
    camera_i_nn_type: none   
    camera_i_ip: '169.254.1.1'
    camera_i_pipeline_type: rgbd
    stereo_i_subpixel: true
    camera_i_enable_imu: true
    camera_i_enable_ir: true
    camera_i_enable_threshold_filter: true
    camera_i_threshold_filter_min_range: 700
    camera_i_threshold_filter_max_range: 4000
    camera_i_restart_on_diagnostics_error: true
```

[image]

export DEPTHAI_DEBUG=1
```
 * /oak_d_pro_poe_front/camera_i_base_frame: oak-d_frame
 * /oak_d_pro_poe_front/camera_i_cam_pitch: 0.015
 * /oak_d_pro_poe_front/camera_i_cam_pos_x: 0.15
 * /oak_d_pro_poe_front/camera_i_cam_pos_y: 0.01
 * /oak_d_pro_poe_front/camera_i_cam_pos_z: 0.0225
 * /oak_d_pro_poe_front/camera_i_cam_roll: 0.0
 * /oak_d_pro_poe_front/camera_i_cam_yaw: -0.005
 * /oak_d_pro_poe_front/camera_i_enable_imu: True
 * /oak_d_pro_poe_front/camera_i_enable_ir: True
 * /oak_d_pro_poe_front/camera_i_enable_threshold_filter: True
 * /oak_d_pro_poe_front/camera_i_imu_from_descr: False
 * /oak_d_pro_poe_front/camera_i_ip: 169.254.1.1
 * /oak_d_pro_poe_front/camera_i_mx_id: 1844301041E8E9F400
 * /oak_d_pro_poe_front/camera_i_nn_type: none
 * /oak_d_pro_poe_front/camera_i_parent_frame: oak-d-base-frame
 * /oak_d_pro_poe_front/camera_i_pipeline_type: rgbd
 * /oak_d_pro_poe_front/camera_i_publish_tf_from_calibration: False
 * /oak_d_pro_poe_front/camera_i_restart_on_diagnostics_error: True
 * /oak_d_pro_poe_front/camera_i_threshold_filter_max_range: 4000
 * /oak_d_pro_poe_front/camera_i_threshold_filter_min_range: 700
 * /oak_d_pro_poe_front/robot_description: <?xml version="1....
 * /oak_d_pro_poe_front/stereo_i_subpixel: True
 * /oak_d_pro_poe_front_depth_image_to_rgb_pointcloud/queue_size: 10
 * /oak_d_pro_poe_left/camera_i_enable_imu: True
 * /oak_d_pro_poe_left/camera_i_enable_ir: True
 * /oak_d_pro_poe_left/camera_i_enable_threshold_filter: True
 * /oak_d_pro_poe_left/camera_i_floodlight_brightness: 200.0
 * /oak_d_pro_poe_left/camera_i_ip: 169.254.3.1
 * /oak_d_pro_poe_left/camera_i_laser_dot_brightness: 200.0
 * /oak_d_pro_poe_left/camera_i_mx_id: 1844301091CBE2F400
 * /oak_d_pro_poe_left/camera_i_nn_type: none
 * /oak_d_pro_poe_left/camera_i_pipeline_type: rgbd
 * /oak_d_pro_poe_left/camera_i_restart_on_diagnostics_error: True
 * /oak_d_pro_poe_left/camera_i_threshold_filter_max_range: 4000
 * /oak_d_pro_poe_left/camera_i_threshold_filter_min_range: 700
 * /oak_d_pro_poe_left/stereo_i_subpixel: True
 * /oak_d_pro_poe_right/camera_i_enable_imu: True
 * /oak_d_pro_poe_right/camera_i_enable_ir: True
 * /oak_d_pro_poe_right/camera_i_enable_threshold_filter: True
 * /oak_d_pro_poe_right/camera_i_floodlight_brightness: 200.0
 * /oak_d_pro_poe_right/camera_i_ip: 169.254.2.1
 * /oak_d_pro_poe_right/camera_i_laser_dot_brightness: 200.0
 * /oak_d_pro_poe_right/camera_i_mx_id: 1844301091FDE9F400
 * /oak_d_pro_poe_right/camera_i_nn_type: none
 * /oak_d_pro_poe_right/camera_i_pipeline_type: rgbd
 * /oak_d_pro_poe_right/camera_i_restart_on_diagnostics_error: True
 * /oak_d_pro_poe_right/camera_i_threshold_filter_max_range: 4000
 * /oak_d_pro_poe_right/camera_i_threshold_filter_min_range: 700
 * /oak_d_pro_poe_right/stereo_i_subpixel: True
 * /rosdistro: noetic
[... 501 lines cut ...]
[1844301041E8E9F400] [169.254.1.1] [34.364] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 0 
[1844301041E8E9F400] [169.254.1.1] [34.496] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 2 
[1844301041E8E9F400] [169.254.1.1] [35.042] [system] [info] Memory Usage - DDR: 168.14 / 333.28 MiB, CMX: 2.10 / 2.50 MiB, LeonOS Heap: 67.95 / 81.76 MiB, LeonRT Heap: 4.58 / 39.90 MiB / NOC ddr: 1008 MB/s
[1844301041E8E9F400] [169.254.1.1] [35.042] [system] [info] Temperatures - Average: 46.28C, CSS: 48.25C, MSS 45.09C, UPA: 45.77C, DSS: 46.00C
[1844301041E8E9F400] [169.254.1.1] [35.042] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 8.64%
[1844301041E8E9F400] [169.254.1.1] [35.101] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 2 
[1844301041E8E9F400] [169.254.1.1] [35.629] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 1 
W: Isp cfg mising: 2 
[1844301041E8E9F400] [169.254.1.1] [35.784] [system] [warning] PRINT:LeonCss: memcpy_dma: needed 2 dst reads
[1844301041E8E9F400] [169.254.1.1] [36.043] [system] [info] Memory Usage - DDR: 168.14 / 333.28 MiB, CMX: 2.10 / 2.50 MiB, LeonOS Heap: 67.95 / 81.76 MiB, LeonRT Heap: 4.58 / 39.90 MiB / NOC ddr: 1010 MB/s
[1844301041E8E9F400] [169.254.1.1] [36.043] [system] [info] Temperatures - Average: 46.34C, CSS: 48.25C, MSS 45.32C, UPA: 46.00C, DSS: 45.77C
[1844301041E8E9F400] [169.254.1.1] [36.043] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 8.48%
[1844301041E8E9F400] [169.254.1.1] [36.795] [system] [warning] PRINT:LeonCss: memcpy_dma: needed 2 dst reads
[1844301041E8E9F400] [169.254.1.1] [36.927] [system] [warning] PRINT:LeonCss: memcpy_dma: needed 2 dst reads
[1844301041E8E9F400] [169.254.1.1] [37.057] [system] [info] Memory Usage - DDR: 168.14 / 333.28 MiB, CMX: 2.10 / 2.50 MiB, LeonOS Heap: 67.95 / 81.76 MiB, LeonRT Heap: 4.58 / 39.90 MiB / NOC ddr: 1010 MB/s
[1844301041E8E9F400] [169.254.1.1] [37.057] [system] [info] Temperatures - Average: 46.45C, CSS: 48.03C, MSS 45.77C, UPA: 45.77C, DSS: 46.23C
[1844301041E8E9F400] [169.254.1.1] [37.057] [system] [info] Cpu Usage - LeonOS 99.99%, LeonRT: 8.42%
^C[oak_d_pro_poe_front_depth_image_to_rgb_pointcloud-6] killing on exit
[oak_d_pro_poe_front-5] killing on exit
[oak_d_pro_poe_front_nodelet_manager-4] killing on exit
[oak_d_pro_poe_front_state_publisher-3] killing on exit
[set_log_level-2] killing on exit
[INFO] [1724279566.639090053]: Unloading nodelet /oak_d_pro_poe_front from manager oak_d_pro_poe_front_nodelet_manager

```

closed 2024-08-22 by Serafadam