luxonis/depthai-ros #572: dynamic config doesn't change anything

khuechuong, 2024-08-06:
So I did dynamic config just to test out calibration for oak-d-pro-poe like https://docs.luxonis.com/software/ros/depthai-ros/driver/ said
problem is the depth image doesn't seem to change at all, for example, I changed the dot brightness:
[image]
[image]

here I changed floodlight brightness
[image]
[image]
I don't see any changes

Serafadam, 2024-08-07:
Hi, are you getting any logs when setting those parameters? Also, could you show view from left/right camera?

khuechuong, 2024-08-07:
I'm not getting logs but I don't think that the problem is dynamic config since the rgb image changes when I check and uncheck rgb_r_set_man_whitebalance. As for the left/right camera, I don't have it since when I run rgbd.launch it only gives me 2 image rostopics: oak/stereo/image_raw and oak/rgb/image_raw

khuechuong, 2024-08-13:
@Serafadam any updates?

Serafadam, 2024-08-14:
Hi, to enable left/right you can edit rgbd.yaml with following values:
```yaml
left_i_publish_topic: true
right_i_publish_topic: true
```
Then those topics should appear, generally just by looking at them (might be better to do in darker environment) you should see if projector dots are affected by config values