luxonis/depthai-ros #575: OAK-D-PRO-POE shutdown after a while

khuechuong, 2024-08-07:
ROS Noetic
OAK-D-PRO-POE

So the oak-d-pro-poe was fine when it starts, but for some reason, after a while it gives me this

```
[1844301041E8E9F400] [169.254.1.221] [1723044660.375] [host] [warning] Monitor thread (device: 1844301041E8E9F400 [169.254.1.221]) - ping was missed, closing the device connection
[ERROR]ros.depthai_ros_driver][produceDiagnostics:80]: No data on logger queue!
[ERROR]ros.depthai_ros_driver][diagCB:61]: Camera diagnostics error: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'sys_logger_queue' (X_LINK_ERROR)'
[ERROR]ros.depthai_ros_driver][produceDiagnostics:80]: No data on logger queue!
[ERROR]ros.depthai_ros_driver][diagCB:61]: Camera diagnostics error: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'sys_logger_queue' (X_LINK_ERROR)'

```

and then shutdown.

Serafadam, 2024-08-08:
Hi, this might be due to high CPU usage on the camera, you can monitor it either via diagnostics topic or by enabling debug information - `export DEPTHAI_DEBUG=1`, you can also set a parameter `camera_i_restart_on_diagnostics_error: true` to have the camera restart after encountering an error

naman-gupta99, 2025-03-01:
> Hi, this might be due to high CPU usage on the camera, you can monitor it either via diagnostics topic or by enabling debug information - `export DEPTHAI_DEBUG=1`, you can also set a parameter `camera_i_restart_on_diagnostics_error: true` to have the camera restart after encountering an error

Hey! What should I look for in the diagnostics topic or the debug logs to find that there is high CPU usage? And even if that is the issue is there no other solution than allowing the camera to keep restarting?

naman-gupta99, 2025-03-05:
@khuechuong Were you able to figure this issue out?
@Serafadam Can you please help?

naman-gupta99, 2025-03-10:
@Serafadam I did add these but I couldn't find anything related to CPU Usage.

```
header:
  stamp:
    sec: 1740806521
    nanosec: 556034201
  frame_id: ''
status:
- level: "\x02"
  name: 'rtabmap: Sync status'
  message: No events recorded.; No data since last update.
  hardware_id: none
  values:
  - key: Events in window
    value: '0'
  - key: Events since startup
    value: '0'
  - key: Duration of window (s)
    value: '4.999979'
  - key: Actual frequency (Hz)
    value: '0.000000'
  - key: Target frequency (Hz)
    value: '0.000000'
  - key: Maximum acceptable frequency (Hz)
    value: '0.000000'
  - key: 'Earliest timestamp delay:'
    value: '0.000000'
  - key: 'Latest timestamp delay:'
    value: '0.000000'
  - key: 'Earliest acceptable timestamp delay:'
    value: '-1.000000'
  - key: 'Latest acceptable timestamp delay:'
    value: '5.000000'
  - key: 'Late diagnostic update count:'
    value: '0'
  - key: 'Early diagnostic update count:'
    value: '0'
  - key: 'Zero seen diagnostic update count:'
    value: '0'
---
header:
  stamp:
    sec: 1740806521
    nanosec: 643101486
  frame_id: ''
status:
- level: "\x02"
  name: 'rtabmap_viz: Sync status'
  message: No events recorded.; No data since last update.
  hardware_id: none
  values:
  - key: Events in window
    value: '0'
  - key: Events since startup
    value: '0'
  - key: Duration of window (s)
    value: '2.999869'
  - key: Actual frequency (Hz)
    value: '0.000000'
  - key: Target frequency (Hz)
    value: '0.000000'
  - key: Maximum acceptable frequency (Hz)
    value: '0.000000'
  - key: 'Earliest timestamp delay:'
    value: '0.000000'
  - key: 'Latest timestamp delay:'
    value: '0.000000'
  - key: 'Earliest acceptable timestamp delay:'
    value: '-1.000000'
[... 213 lines cut ...]
    value: '4.999944'
  - key: Actual frequency (Hz)
    value: '0.000000'
  - key: Target frequency (Hz)
    value: '0.000000'
  - key: Maximum acceptable frequency (Hz)
    value: '0.000000'
  - key: 'Earliest timestamp delay:'
    value: '0.000000'
  - key: 'Latest timestamp delay:'
    value: '0.000000'
  - key: 'Earliest acceptable timestamp delay:'
    value: '-1.000000'
  - key: 'Latest acceptable timestamp delay:'
    value: '5.000000'
  - key: 'Late diagnostic update count:'
    value: '0'
  - key: 'Early diagnostic update count:'
    value: '0'
  - key: 'Zero seen diagnostic update count:'
    value: '0'
---
```

And I have the OAK-D Lite Camera.

naman-gupta99, 2025-03-12:
@Serafadam Can you please check this and help?

Serafadam, 2025-03-12:
Hi, it seems that the camera encounters some errors and shuts down, could you run 
`export DEPTHAI_DEBUG=1` before running the driver and paste the logs?

naman-gupta99, 2025-03-13:
Hello @Serafadam.
Thank you for your response. You can see the log file here.

[logfile.txt](https://github.com/user-attachments/files/19220495/logfile.txt)

Serafadam, 2025-03-13:
Thanks for the report, it looks like the device is running fine but it seems that there is some issue related to diagnostics which  could be firmware related. Could you disable diagnostics? You can do that by setting `pipeline_gen.i_enable_diagnostics: false` parameter

naman-gupta99, 2025-04-16:
Hi @Serafadam,  
Apologies for the delayed response.

I made the following change to the `ugv_ws/src/ugv_main/ugv_vision/config/oak_d_lite.yaml` file:

```yaml
/oak:
  ros__parameters:
    camera:
      i_enable_imu: false
      i_enable_ir: false
      i_nn_type: none
      i_enable_diagnostics: false
    stereo:
      i_subpixel: true
```

Could you please confirm if this is the correct location for these parameters? I understand this setup may differ from yours since it's running on a different platform, so no worries if it's outside the scope of what you can answer.

That said, even with these changes, the camera still turns off after a short time.

naman-gupta99, 2025-04-22:
@Serafadam Can you please check this?