luxonis/depthai-ros #577: low horizontal and vertical FOV

khuechuong, 2024-08-08:
ROS Noetic, rgbd_pcl.launch, Oak-D-Pro-POE-FF

So I wrote a code for a frostum that represents the FOV of oak-d-pro-ff-poe using the horizontal and vertical angle:

```
def frostum_vertices(near, far, h_fov, v_fov):
    """
    Calculate vertices of a frustum pyramid in 3D space based on given parameters.

    Parameters:
    near (float): Distance to the near plane of the frustum.
    far (float): Distance to the far plane of the frustum.
    h_fov (float): Horizontal field of view (in degrees).
    v_fov (float): Vertical field of view (in degrees).
    plot (bool, optional): Whether to plot the vertices in a 3D space. Default is False.

    Returns:
    ndarray: Array of shape (8, 3) containing the vertices of the frustum pyramid in local coordinates.
    """
    # convert from degree to rad
    h_fov_rad = np.radians(h_fov)
    v_fov_rad = np.radians(v_fov)
    
    # Calculate top and bottom near plane dimensions
    x_near = near 
    y_near = near *np.tan(h_fov_rad/2)
    z_near = near * np.tan(v_fov_rad / 2)

    x_far = far 
    y_far = far *np.tan(h_fov_rad/2)
    z_far = far * np.tan(v_fov_rad / 2)
    
    # Define the vertices of the frustum pyramid in local coordinates
    vertices = np.array([
        [x_near,  y_near, -z_near],     # Bottom left corner of near plane
        [x_near, -y_near, -z_near],     # Bottom right corner of near plane
        [x_near, -y_near,  z_near],     # Top right corner of near plane
        [x_near,  y_near,  z_near],     # Top left corner of near plane
        [x_far,  y_far, -z_far],        # Bottom left corner of far plane
        [x_far, -y_far, -z_far],        # Bottom right corner of far plane
        [x_far, -y_far,  z_far],        # Top right corner of far plane
        [x_far,  y_far,  z_far],        # Top left corner of far plane
    ])
```

and 

```
marker = Marker()
    marker.header.frame_id = "oak"  # Replace with your frame_id
    marker.type = Marker.TRIANGLE_LIST
    marker.action = Marker.ADD
    marker.scale.x = 1.0
    marker.scale.y = 1.0
    marker.scale.z = 1.0
    marker.color.r = 0.0
    marker.color.g = 1.0
    marker.color.b = 0.0
    marker.color.a = 0.5  # Adjust transparency as needed

    # Define the triangles for the frustum faces
    faces = [
        # Near plane
        (0, 1, 2),
        (0, 2, 3),
        # Far plane
        (4, 5, 6),
        (4, 6, 7),
        # Side faces
        (0, 1, 5),
        (0, 5, 4),
        (1, 2, 6),
        (1, 6, 5),
        (2, 3, 7),
        (2, 7, 6),
        (3, 0, 4),
        (3, 4, 7),
    ]
    
    # Add vertices to Marker as triangles
    for face in faces:
        for i in face:
            p = Point()
            p.x, p.y, p.z = vertices[i]
            marker.points.append(p)
    # Publish Marker
    pub_marker.publish(marker)
```
it's not exactly the same, but I think it's close enough. From what I saw in the specs, RGB fixed focus image has a 69 h fov/ 55 v fov respectively.  However, as seen here:

[image]

when I had that marker in rviz, I had to lower it down to 62 and 37.5 just for the it to overlap a little. You notice there are a little bit sticking out of frostum. If I change it to 65/38, nothing would stick out. Is there a reason why the FOV decreases so much?

Serafadam, 2024-08-13:
Hi, I think this is due to aligning process, for the PCL to be aligned properly both image planes need to encompass the same area, stereo image needs to be created from two images that need to be rectified which might result in a smaller FOV for both of them, and then that rectified image needs to be aligned to RGB frame which might introduce additional crop depending on the sensor. If you want to get larger FOV for stereo you can increase alpha scaling parameter accordingly, which should result in larger FOV.

khuechuong, 2024-08-13:
so set ``` i_enable_alpha_scaling: true```? like that? I just see a boolean. Is there another parameter can I increase to increase to FOV or is it just set it to true?

Serafadam, 2024-08-14:
Hi, to set specific value for alpha you can use `i_alpha_scaling` parameter in stereo, the values are from 0 to 1.0. Overall, you can check what parameters are available and how they are set [here](https://github.com/luxonis/depthai-ros/blob/82ce67cf91182b11b26504af3973dc312824970e/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp#L124) in the code, as for their detailed description you can refer to [official documentation](https://docs.luxonis.com/software/depthai-components/nodes/stereo_depth/)