luxonis/depthai-ros #584: [BUG] {X_LINK_DEVICE_NOT_FOUND inside depthai-ros container}

Aadi0902, 2024-08-23:
When running the depthai_viewer using python3 -m depthai_viewer inside deptha-ai ros container (ros humble), I keep getting

```
Failed to select device: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

[2024-08-23T06:48:01Z ERROR re_viewer::depthai::depthai] Error: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND, Try to connect the device to a different port.

[2024-08-23 06:48:02.166] [depthai] [warning] skipping X_LINK_UNBOOTED device having name "1.12.4.4"
```

What is weird to me is that my udev rules are set on the host, and I am mounting my /dev/usb folder.

Additionally, the viewer works fine on my host and I am able to view my camera output, but inside the container I am getting the above error Please help.

If curious, this was my docker run command:

`docker run -it --privileged -v /dev/bus/usb:/dev/bus/usb -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix luxonis/depthai-ros:humble-latest /bin/bash`

Serafadam, 2024-08-23:
Hi, which device are you using? Are results of running `lsusb` command the same on host and in docker?

Aadi0902, 2024-08-24:
I am using oak d pro. And yes, I can confirm `lsusb` has the same output inside and outside the container. I can see `Bus 001 Device 047: ID 03e7:2485 Movidius Ltd. Movidius MyriadX` on both.

sourenpash, 2025-12-10:
same issue