luxonis/depthai-ros #591: [BUG] {Title of the bug} drop in pointcloud rate after running for a couple of hours

kushal2507, 2024-09-05:
**Describe the bug**
I am running 2 OAK D S2 on a 13th Gen Intel® Core™ i7-13700HX × 24 with 16 GB ram, ROS2 Humble and the latest code from the humble branch. . The image and pointcloud rate of publish are fine at the beginning (around 27-30 HZ) but after an hour or so, the pointcloud for both cameras drops to 1hz or less whereas the rgb images and depth images are fine at 27-30hz. 

I have also tried running it on depth_image_proc::PointCloudXyzrgbNode instead of depth_image_proc::PointCloudXyzNode(see launch file below), but the performance seems to be worse there as the poitncloud is only published at 6-7 hz and eventually drops to less than 1.

**Minimal Reproducible Example**
Here is my launch file to launch the multiple cameras : 

`

    import os

    from ament_index_python.packages import get_package_share_directory
    from launch import LaunchDescription
    from launch_ros.actions import LoadComposableNodes
    from launch_ros.descriptions import ComposableNode
    from launch.actions import (
     IncludeLaunchDescription,
     OpaqueFunction,
     )
     from launch.launch_description_sources import PythonLaunchDescriptionSource

    def launch_setup(context, *args, **kwargs):

        config_prefix = get_package_share_directory("bgl_hardware_launcher")
        params_file = os.path.join(config_prefix, "config", "bgl_multi_cam.yaml")
        cams = ["camera_front", "camera_rear"]
        nodes = []
        for cam_name in cams:
            rgb_topic_name = cam_name + "/rgb/image_raw"
            camera_launch = IncludeLaunchDescription(
                PythonLaunchDescriptionSource(
                    os.path.join(config_prefix, "launch", "camera.launch.py")
                ),
                launch_arguments={
                    "name": cam_name,
                    "parent_frame": "base_link",
                    "params_file": params_file,
                }.items(),
            )
            nodes.append(camera_launch)

            pointcloud_node = LoadComposableNodes(
                target_container=cam_name + "_container",
                composable_node_descriptions=[
                    ComposableNode(
                        package="depth_image_proc",
                        plugin="depth_image_proc::PointCloudXyzNode",
                        name=cam_name + "point_cloud_xyz_node",
                        remappings=[
                            ("image_rect", cam_name + "/stereo/image_raw"),
                            ("camera_info", cam_name + "/stereo/camera_info"),
                            ("points", cam_name + "/points"),
                        ],
                    ),
                ],
            )
            nodes.append(pointcloud_node)

        return nodes

    def generate_launch_description():

        return LaunchDescription(
            [
                OpaqueFunction(function=launch_setup),
            ]
       )
`
Here is the yaml file : 

`

    /camera_front:
      ros__parameters:
      camera:
        i_mx_id: 19443010E106191300
        i_nn_type: none
        i_enable_imu: false
        i_pipeline_type: RGBD
        i_restart_on_diagnostics_error: true
      rgb:
        i_fps: 30.0
        i_resolution: 1080P
      stereo:
        i_fps: 30.0
        i_subpixel: true
      
    /camera_back:
    ros__parameters:
      camera:
        i_mx_id: 19443010817F1D1300
        i_nn_type: none
        i_enable_imu: false
        i_pipeline_type: RGBD
        i_restart_on_diagnostics_error: true
      rgb:
        i_fps: 30.0
        i_resolution: 1080P
      stereo:
        i_fps: 30.0
        i_subpixel: true
`

**Expected behavior**
I expect the pointcloud to continue publishing at same rate as long as the camera keeps running.

**Screenshots**
[image]

This is the output from /diagnostics in case it is useful.

**Pipeline Graph**

Front camera pipeline - 
[19443010E106191300_pipeline.json](https://github.com/user-attachments/files/16884476/19443010E106191300_pipeline.json)

Rear camera pipeline - 
[19443010817F1D1300_pipeline.json](https://github.com/user-attachments/files/16884479/19443010817F1D1300_pipeline.json)

Serafadam, 2024-09-16:
Hi, sorry for the delay, could you run the code with [pipeline_graph](https://github.com/luxonis/depthai_pipeline_graph) tool? You can run it as following:
`pipeline_graph run "your_launchfile"`.
If the drops are happening on the pipeline level then it might be on the device side but those issues could be caused by DDS.

kushal2507, 2024-09-17:
Hi @Serafadam thank you for your reply. I ran it with the tool and I have attached the results below (json dump and image) . In the graph, I do not see a node for "pointcloud"? Not sure if that is expected? Another thing to point out is that on the tool, the FPS does not drop below 58 (for both cameras) but upon doing ` ros2 topic hz [topic name] ` the rate published is usually nowhere close to the 30 being shown on the pipeline, it is usually around 20, so I would expect the FPS to be around 40-45. 
Please let me know if you need some other information.
Thanks

Note :  I am running the default `rmw-fastrtps-cpp` DDS without any tuning or modifications.
[image]
[pipeline_graph.json](https://github.com/user-attachments/files/17024350/pipeline_graph.json)

Serafadam, 2024-09-17:
Hi, `ros2 topic hz` can be actually misleading as it is written in python (and not composable) and will introduce additional overhead if used on larger messages, you can try checking the frequency of `camera_info` or `compressed` topics, this could provide better results. From the graph you posted the issue is not on the device's side, which suggests that the issue is ROS related. You could try [DDS tuning methods](https://docs.ros.org/en/jazzy/How-To-Guides/DDS-tuning.html). You could also check if the issue still persists after removing PCL node, if yes then the issue might be with this node.

kushal2507, 2024-10-01:
The issue I beleive was from the way I was creating the composable container. If I use different containers for the pointcloud node  and rgb node, the pointcloud remains stable at 25-30hz (throttles a bit sometimes but comes back to 30 once the depth image also starts publishing at 30). Here is my updated launch file : 

    import os
    
    from ament_index_python.packages import get_package_share_directory
    from launch import LaunchDescription
    from launch_ros.actions import LoadComposableNodes
    from launch_ros.descriptions import ComposableNode
    from launch_ros.actions import ComposableNodeContainer
    from launch.actions import (
        IncludeLaunchDescription,
        OpaqueFunction,
    )
    from launch.launch_description_sources import PythonLaunchDescriptionSource
    
    
    def launch_setup(context, *args, **kwargs):

      config_prefix = get_package_share_directory("bgl_hardware_launcher")
      params_file = os.path.join(config_prefix, "config", "bgl_multi_cam.yaml")
      cams = ["camera_front", "camera_rear"]
      nodes = []
      for cam_name in cams:
          depth_image_topic = "/depth/image_raw"
          depth_image_camera_info_topic = "/depth/camera_info"
          camera_launch = IncludeLaunchDescription(
              PythonLaunchDescriptionSource(
                  os.path.join(config_prefix, "launch", "camera.launch.py")
              ),
              launch_arguments={
                  "name": cam_name,
                  "parent_frame": "base_link",
                  "params_file": params_file,
              }.items(),
          )
          nodes.append(camera_launch)
  
          # Create a composable node container for each camera
          container = ComposableNodeContainer(
              name=cam_name + "_container",
              namespace=cam_name,
              package="rclcpp_components",
              executable="component_container",
              output="screen",
          )
          nodes.append(container)
  
          pointcloud_node = LoadComposableNodes(
              target_container=container,
              composable_node_descriptions=[
                  ComposableNode(
                      package="depth_image_proc",
                      plugin="depth_image_proc::PointCloudXyzNode",
                      name=cam_name + "point_cloud_xyz_node",
                      remappings=[
                          ("image_rect", cam_name + depth_image_topic),
                          ("camera_info", cam_name + depth_image_camera_info_topic),
                          ("points", cam_name + "/points"),
                      ],
                  ),
              ],
          )
          nodes.append(pointcloud_node)
  
      return nodes
 
    def generate_launch_description():
  
        return LaunchDescription(
            [
                OpaqueFunction(function=launch_setup),
            ]
        )

bartonp2, 2025-01-15:
I had the same issue with the increase in delay and decrease in frequency after an hour or so.
Delay: 0.1s -> 4s
Frequency: 30Hz -> 1.0Hz (0.001 

Running the pointcloud node in its own composable container also fixed the issue for me. However, I am left wondering if something is wrong with the ros configuration/setup for intraprocess communication, or if the problem originates from the node?