luxonis/depthai-ros #613: [Feature-Request] Support autoexposure with maximum exposure duration

abaeyens-imod, 2024-10-24:
### `why`
We and our clients use the Luxonis cameras on moving robots for VIO and VSLAM purposes. When we currently use the Luxonis cameras in dark environments, the exposure duration becomes too long resulting in severe motion blur, making the resulting images unusable (or at least result in severely degraded performance). Setting the exposure manually provides usable images for that lighting environment, but doesn't work in other environments.

Some other cameras, such as the Intel RealSense T265, don't offer this as a setting but do this automatically (presumably they have a suitable maximum exposure duration hardcoded).

### `what`
- Add a parameter `max_exposure` that sets a maximum exposure duration.
- Autoexposure works as before, except that the ISO/sensitivity is increased as necessary to keep the exposure duration lower than `max_exposure`. It's fine if the image becomes noisy, and dark once the sensitivity has reached its maximum.
- This is most important for the stereo pair, though also useful for the RGB camera.
- In case it's not possible to make this adjustable, and that only one particular maximum exposure duration is supported, we're open to discussing what value would be suitable.

### `how`
No idea, not familiar with the exact possibilities of these cameras.

Serafadam, 2024-10-25:
Hi, please check [this branch](https://github.com/luxonis/depthai-ros/tree/noetic_additional_camera_control) to see if this implementation works for your case. You can find the new parameters [here](https://github.com/luxonis/depthai-ros/commit/9a2e0433446c08d37d424d3b6caa463eb9c3a5bb)

abaeyens-imod, 2024-10-25:
Really appreciated, I'll test and report back early next week.

abaeyens-imod, 2024-10-29:
Tested together with https://github.com/luxonis/depthai-ros/issues/614, works well.