luxonis/depthai-ros #614: [Feature-Request] expose sharpness setting in the ROS driver

abaeyens-imod, 2024-10-24:
### `why`
When currently using the Luxonis cameras with the ROS driver, the default sharpness is set quite high and not adjustable. We would prefer that it can be set lower to avoid the current oversharpening artefacts and the amplified image noise.

### `what`
- A new parameter `sharpness` that adjusts the sharpening.
- This parameter doesn't have to be dynamic.
- We would appreciate seeing this also for Noetic.

### `how`
At first sight: this setting is already available in the C++ API, so it's mostly a matter of exposing it in the ROS driver.

### also
If possible, also expose the parameters `luma_denoise` and `chroma_denoise`.

Serafadam, 2024-10-25:
Hi, please check [this branch](https://github.com/luxonis/depthai-ros/tree/noetic_additional_camera_control) to see if this implementation works for your case. You can find the new parameters [here](https://github.com/luxonis/depthai-ros/commit/9a2e0433446c08d37d424d3b6caa463eb9c3a5bb)

abaeyens-imod, 2024-10-29:
@Serafadam just tested this together with https://github.com/luxonis/depthai-ros/issues/613, it works great and makes the Luxonis cameras considerably more usable for VSLAM. Thank you for addressing this swiftly. Two nits:
- `i_set_chroma_denoise` works well for the RGB camera, but I'm not sure what it's supposed to do for the mono cameras.
- the new parameters don't show up in `rqt_reconfigure` yet, I assume related to that they aren't yet listed in `depthai_ros_driver/cfg/parameters.cfg`.

Serafadam, 2024-10-30:
Hi, please check the newest version, params have been changed to runtime. Regarding chroma denoise, I think it applies mostly in the cases where stereo pair is color, like OAK-D SR

abaeyens-imod, 2024-11-04:
Tested, the dynamic parameters work well. For evaluating denoise and sharpening settings it's quite useful. (I now notice I didn't ask for dynamic parameters, so normal that I didn't see them in rqt_reconfigure :see_no_evil:)

Concerning the allowable range, in [the Luxonis API doc](https://docs.luxonis.com/software/depthai-components/messages/camera_control/#depthai.CameraControl.setLumaDenoise) mentions that the range for `setLumaDenoise` is 0..4 while rqt_reconfigure only allows 1..4 (screenshot below). That feels contradictory, is this desired? Apart from that, good to merge as far as I'm concerned.
[image]

Serafadam, 2024-11-06:
Hi, thanks for noticing, the ranges should be updated in the latest commit

Interpause, 2025-01-20:
@Serafadam thanks for your work. Could I ask when the same changes will be ported to the humble branch? thanks

Serafadam, 2025-01-21:
Hi, those changes [are already on Humble branch](https://github.com/luxonis/depthai-ros/blob/cf71d9d2ad1a08bb39127ab5303f94f016e95a4d/depthai_ros_driver/src/param_handlers/sensor_param_handler.cpp#L83)