luxonis/depthai-ros #615: Very noisy map using Rtabmap

mizrex, 2024-10-24:
We have tried using Rtabmap with Oak-D-Pro W and Oak-D-Pro-W-POE, both give out very noisy maps. I have tried tuning various parameters within Rtabmap. But it looks like the data from the cameras are very noisy. Are there any parameters within the camera that needs tuning?
Below are samples of maps that we were able to generate. You can see, how the open space around the unit is interpreted as obstacles.
[image]
[image]

Serafadam, 2024-10-25:
Hi, judging by those maps I would say the issue could be related to the default RTABMap's parameters, I would advise investigating ground/obstacle pointclouds with either rviz or RTABMap's database viewer. As for the parameters themselves, I would start with setting  `Grid/RayTracing=true` and `Grid/3D=false` and checking the obstacle height parameter. Is the camera attached to a robot, or in handheld mode?

mizrex, 2024-10-25:
@Serafadam , thank you for getting back.
The camera is mounted to a wheeled robot, about 0.05 m from the ground. 
I'll try out `Grid/RayTracing=true` and `Grid/3D=false`, and update here shortly.
I had looked up the ground/obstacle and pointclouds. It looks like the odom/orientation is messed up. I'm not sure if it's an IMU issue or Visual odom in Rtabmap. You can see the Rviz screenshots here, how the odom is pointing into the ground.
[image]
[image]
[image]

mizrex, 2024-10-25:
@Serafadam , you are the best. Setting `Grid/RayTracing=true` and `Grid/3D=false` seem to have worked, and improved it quite a bit. However, there's still some discrepancy and noise in the data. Any other parameters you would suggest trying?
Here are the revised maps.
You can notice, how there are darker and thicker black regions (as if the camera can see through the obstacles), and the odom is pointing into the ground (at an angle), even though the camera is pointing straight, parallel to the ground.
[image]
[image]