luxonis/depthai-ros #616: how to publish stereo rectified color image for left and right camera in 400P?

lturing, 2024-10-29:
hi, I have OAK-D-SR, which has camera OV9782 and support rgb.       
the default rectified stereo publish gray-scale image for left and right (depthai_examples/src/stereo_inertial_publisher.cpp).    
for my application, i need the rectified stereo for slam and rgb-scale for deep learning.

Serafadam, 2024-10-30:
Hi, I would suggest adding using `depthai_ros_driver` for that as you can just set `left/right.i_publish_topic: true` which should give you color streams, and then use rectify node in ROS to remove distortion, similarly to how it's done [here](https://github.com/luxonis/depthai-ros/blob/9132443a85381f118e75d8583df3d9e764a91b7a/depthai_ros_driver/launch/camera.launch.py#L226)

lturing, 2024-10-30:
Thank you for your reply.     
i use the method you mentioned, for my application, the left/right color images and stereo rectified left/right image's  timestamps need be the same, the method you mentioned publish these separately, and cannot guarante synchronization of left and right image timestamps.  
it will better to simultaneously publish rectified color left/right image, thank you a lot.

lturing, 2024-10-30:
@Serafadam  Thank you for your reply.
i use the method you mentioned, for my application, the left/right color images and stereo rectified left/right image's timestamps need be the same, the method you mentioned publish these separately, and cannot guarante synchronization of left and right image timestamps.
it will better to simultaneously publish rectified color left/right image, thank you a lot.

lturing, 2024-11-06:
@Serafadam 
i use depthai_ros_driver/launch/camera.launch.py to publish raw left/right image, and use "image_proc::RectifyNode" to rectify them, the timestamp is followed:       

```
[INFO] [1730880007.262391017] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.207883
[INFO] [1730880007.263250313] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.207876
[INFO] [1730880007.288508682] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.257876
[INFO] [1730880007.289215117] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.257883
[INFO] [1730880007.339837535] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.307876
[INFO] [1730880007.344171716] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.307886
[INFO] [1730880007.389211416] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.357883
[INFO] [1730880007.389900147] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.357876
[INFO] [1730880007.444667497] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.407885
[INFO] [1730880007.445348543] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.407876
[INFO] [1730880007.489026490] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.457883
[INFO] [1730880007.489738916] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.457876
[INFO] [1730880007.540102982] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.507876
[INFO] [1730880007.540789139] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.507883
[INFO] [1730880007.590778281] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.557876
[INFO] [1730880007.595876277] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.557885
[INFO] [1730880007.639657219] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.607885
[INFO] [1730880007.640347654] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.607876
[INFO] [1730880007.689156439] [camera_info_subscriber]: Left Camera Timestamp (seconds): 1730880007.657876
[INFO] [1730880007.693225015] [camera_info_subscriber]: Right Camera Timestamp (seconds): 1730880007.657884
```

it meet the requirements of time synchronization between left and right images. many thanks.

closed 2024-11-06 by lturing