luxonis/depthai-ros #619: [BUG] tf_static published twice

abaeyens-imod, 2024-11-04:
**Describe the bug**
When setting `publish_tf_from_calibration` to true, several frame transformations (such as T_`oak_right_camera_frame`_`oak_left_camera_frame`) get published twice. Not sure what exactly happens, it seems first a nominal one (when launching the driver) and then one from calibration (once the camera connection is established).
This causes an issue because many ROS nodes do not expect camera calibrations published on `tf_static` to be updated dynamically, resulting in errors and/or the extrinsics from calibration going unused. Also, in some cases (see example below), the tree structure also changes, which results in some frames having two parent frames.

Example output of `rostopic echo /tf_static` (part of the output has been replaced with `...` for brevity):
```
---
...
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1730742195
        nsecs: 106724901
      frame_id: "oak"
    child_frame_id: "oak_left_camera_frame"
    transform: 
      translation: 
        x: 0.0
        y: 0.0375
        z: 0.0
      rotation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
... 
---
...
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1730742205
        nsecs: 734899265
      frame_id: "oak_right_camera_frame"
    child_frame_id: "oak_left_camera_frame"
    transform: 
      translation: 
        x: 0.00021379999816417694
        y: 0.07467092037200927
        z: -0.00026392366737127305
      rotation: 
        x: 0.0002737994571999808
        y: -0.002343944984574956
        z: 1.7709348192618804e-05
        w: 0.9999972134789744
...
```
**Minimal Reproducible Example**
Launch file:
```xml
<launch>
  <include file="$(find depthai_ros_driver)/launch/camera.launch">
    <arg name="params_file" value="$(find put_package_name_here)/config/camera.yaml" />
    <param name="publish_tf_from_calibration" value="true" />
  </include>
</launch>
```
Config `camera.yaml`:
```yaml
/oak:
  camera_i_enable_imu: true
  camera_i_publish_tf_from_calibration: true
  camera_i_tf_imu_from_descr: false
...
```
I assume this is the intended way to launch the camera, but if it's just me doing something wrong here, please let know.

**Expected behavior**
- Each frame transformation gets published only once on `tf_static`.
- If `camera_i_publish_tf_from_calibration` is true, then no messages are published on `tf_static` before the calibration is read from the camera.

**Screenshots**
None.

**Pipeline Graph**
Standard pipeline `RGBStereo`.

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
 - Which OS/OS version are you using? **Ubuntu 20.04/Focal container on Ubuntu 22.04/Jammy**
 - Which ROS version are you using? **ROS 1**
 - Which ROS distribution are you using ? **Noetic**
 - Is `depthai-ros` built from source or installed from apt? **apt**
 - Is `depthai/depthai-core` library installed from rosdep or manually? **Installed from apt as part of `ros-noetic-depthai-ros`**
 
**Versions**
 
| package | version |
| -- | -- |
| ros-noetic-depthai | 2.28.0-1focal.20240822.150859 |
| ros-noetic-depthai-ros | 2.10.3-1focal.20241015.133934 |
| ros-noetic-depthai-bridge | 2.10.3-1focal.20241015.132038 |
| ros-noetic-depthai-ros-msgs | 2.10.3-1focal.20241015.131422 |
| ros-noetic-depthai-ros-driver | 2.10.3-1focal.20241015.132927 |

**Additional context**
Please let know if there's anything unclear.

Serafadam, 2024-11-07:
Hi, do you get the same behavior by just running `roslaunch depthai_ros_driver camera.launch publish_tf_from_calibration:=true`? `publish_tf_from_calibration` here is an argument, not a parameter and based on it it is decided whether TF publisher from `depthai_descriptions` is launched

abaeyens-imod, 2024-11-07:
> `publish_tf_from_calibration` here is an argument, not a parameter and based on it it is decided whether TF publisher from depthai_descriptions is launched

You're correct, that explains the weird behavior. Setting it as an argument resolves the issue indeed.

tl;dr: `camera.launch` has a launch file argument `publish_tf_from_calibration` that (1) disables the nominal TF publisher and (2) sets the parameter `camera_i_publish_tf_from_calibration` such that the camera node will query the camera for its calibration.

closed 2024-11-07 by abaeyens-imod