luxonis/depthai-ros #622: [Feature-Request] { using Docker with the ROS2 Jazzy image.}

sfeng97, 2024-11-08:
"I am using Docker with the ROS2 Jazzy image. I downloaded your package, but during compilation, I got the error:

--- stderr: depthai_examples
CMake Error at CMakeLists.txt:41 (file):
file DOWNLOAD cannot compute hash on failed download
status: [28;"Timeout was reached"]

What could be the cause of this?"

sfeng97, 2024-11-08:
[image]

Serafadam, 2024-11-08:
Hi, do you have access to internet connection on that machine? `depthai_examples` download NN blobs while building so that issue is related to download, either by network issue (connection or server error)