luxonis/depthai-ros #623: [BUG] /oak/nn/detections are not published

fzoric8, 2024-11-08:
Hi, 

I'm running `example_marker_publish.launch`, and `obj_det.py` is not starting. 

I've narrowed the problem down to the fact that `/oak/nn/detections` are not published. 

However, when I run `camera.launch`, I get following output: 
```
[ INFO] [1731066454.567315064]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1731066454.572465416]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
process[publisher-7]: started with pid [70059]
[ INFO] [1731066454.585930445]: Initializing nodelet with 8 worker threads.
[ INFO] [1731066454.594745228]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ INFO] [1731066454.821202883]: Starting camera.
[ INFO] [1731066454.827847091]: No ip/mxid/usb_id specified, connecting to the next available device.
[ INFO] [1731066457.690112831]: Camera with MXID: 1944301031529A2E00 and Name: 1.2.4 connected!
[ INFO] [1731066457.691312185]: USB SPEED: HIGH
[ INFO] [1731066457.743272993]: Device type: OAK-D-PRO-W
[ INFO] [1731066457.752503337]: Pipeline type: rgbd
[ INFO] [1731066458.369109780]: NN Family: YOLO
[ INFO] [1731066458.460043851]: NN input size: 416 x 416. Resizing input image in case of different dimensions.
[ INFO] [1731066458.806491628]: Finished setting up pipeline.
``` 

Environment: 
- Ubuntu 20.04 
- ROS noetic 
- docker isolation 

In `depthai-viewer` I can see detections when I use `yolo` and `mobilenet`. But when I use default models from the roslaunch, such as `mobilenet`, and `yolo`, however, after setting pipeline. 

Can you please point me to the right direction, because I'm not sure what to change/setup :) 

Thanks

Serafadam, 2024-11-08:
Hi, could you check if examples in `depthai_filters` work for you? `obj_det.py` has been deprecated and will be removed in future releases

fzoric8, 2024-11-11:
Hi @Serafadam, 

Examples in the `depthai_filters` and `depthai_examples` also work nicely. 

I can use that. I will remap topics, etc.. but yeah, it serves my purpose. 

Thanks for the fast response!

closed 2024-11-11 by fzoric8