luxonis/depthai-ros #628: [Feature-Request] Missing standard (?) frame camera_depth_optical_frame

flabrosse, 2024-11-15:
rtabmap requires the frame `camera_depth_optical_frame` as the frame for the depth image.  There seems to be no way to configure this in rtabmap.  Other cameras provide this frame, such as the realsense, the gazebo kinect driver and probably the ZED (although I cannot check this right now).

It is obviously easy to add a static transform from a provided frame to the needed one, but it would be even better integration if the depthai-ros driver would conform to what seems to be a standard naming of camera frames.  There are other frames that seem to be standard, such as `camera_color_optical_frame` and `camera_imu_optical_frame`.

Serafadam, 2024-11-19:
Hi, 
1. Unless I'm wrong there shouldn't be a requirement for a specific frame name, `rtabmap.launch.py` works with RGBD frames and doesn't throw any errors regarding them, the most important frame is the base_frame which can be overriden.
2. Unfortunately there is no REP (yet) for specifying the frame names (not counting `optical_frame`s)
3. If you enable RS mode (more information in the docs), the topic and frame names will be remapped to match those on Realsense

flabrosse, 2024-11-20:
I found what the issue is.

With the camera simulated by gazebo, the frame_ID is set to the non-existent `camera_depth_optical_frame`:
```
---
header:
  stamp:
    sec: 190
    nanosec: 447000000
  frame_id: camera_depth_optical_frame
```
The frame given by the real camera is `oak_rgb_camera_optical_frame`:
```
header:
  stamp:
    sec: 1732099885
    nanosec: 990851146
  frame_id: oak_rgb_camera_optical_frame
```