luxonis/depthai-ros #630: ros2 jazzy turtlebot4 has not depth topics

orangelee89, 2024-11-21:
ros2 : jazzy
device : oakd-pro-w

when I run 
**ros2 launch turtlebot4_bringup lite.launch.py** 
or
** ros2 launch turtlebot4_bringup oakd.launch.py ** 
 I got topics without depth
[image]

in the ttbot4 launch file 

but wen I run 
**ros2 launch turtlebot4_bringup oakd_example.launch.py** 

 **ros2 launch depthai_examples stereo_inertial_node.launch.py** 
I'm able to get depth topic 
[image]

orangelee89, 2024-11-22:
since I change [turtlebot4_bringup/config/oakd_pro.yaml]( https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/config/oakd_pro.yaml) as below 
[image]
I got the result like:
[image]

Serafadam, 2024-11-25:
Hi, it seems there is an issue with connection, you can enable additional logs by running `export DEPTHAI_DEBUG=1`

orangelee89, 2025-01-01:
hi, now I have other question that there is a big delay on /ttbot1/oakd/rgb/image_raw topic when I saw the topic in rviz2

orangelee89, 2025-01-01:
but when I use /ttbot1/oakd/rgb/preview/image_raw ,it’s showing real time

orangelee89, 2025-01-01:
[image]
and When I use ros2 , which RGB and Depth image is recommended to use

orangelee89, 2025-01-01:
since I change [turtlebot4_bringup/config/oakd_pro.yaml](https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/config/oakd_pro.yaml) as below
(    i_enable_preview: true. change to false)
[image]

the raw rgb iamge topic is batter ,but , I can’t the /stereo/image_raw topic