luxonis/depthai-ros #633: Issues with Obtaining Orientation Values from /oak/imu/data on OAK-D-W using BNO085

dbparkJ, 2024-11-26:
Hello,

I am currently working with the OAK-D-W platform and attempting to obtain orientation values from the `/oak/imu/data` topic. My setup includes the BNO085 sensor. However, I noticed that the orientation field in the `/oak/imu/data` topic always outputs as follows:

```
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

This result indicates that no valid orientation data is being provided, despite the following configuration in my `camera.yaml` file:

```
/oak:
  camera_i_enable_imu: true
  camera_i_enable_ir: true
  camera_i_floodlight_brightness: 0.0
  camera_i_ip: ''
  camera_i_laser_dot_brightness: 800.0
  camera_i_mx_id: ''
  camera_i_nn_type: spatial
  camera_i_pipeline_type: rgbd
  camera_i_usb_port_id: ''
  camera_i_usb_speed: SUPER_PLUS
  nn_i_nn_config_path: depthai_ros_driver/mobilenet
  imu_i_enable_rotation: true
  imu_i_message_type: IMU
```

Could you please clarify the following:
 1. Is the BNO085 sensor capable of providing orientation values (e.g., `ROTATION_VECTOR`) on the `/oak/imu/data` topic?
 2. Are there additional configurations required to enable valid orientation values?
 3. Is this issue related to a limitation with the BNO085 sensor on the DepthAI platform?

Thank you for your support.

Serafadam, 2024-11-26:
Hi, with those parameters you should be able to obtain those values, could you test if this parameters are loaded correctly? You can get full log by running `export DEPTHAI_DEBUG=1` before running the launch file

dbparkJ, 2024-11-27:
Thank you for explaining the method.
The values obtained after running `export DEPTAI_DEBUG=1` are as follows.

### DEPTAI_DEBUG=1 log
```
... logging to /root/.ros/log/17d73064-ac52-11ef-9eb1-02428e0b7472/roslaunch-jm-desktop-724.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jm-desktop:35723/

SUMMARY
========

PARAMETERS
 * /oak/camera/i_enable_imu: True
 * /oak/camera/i_enable_ir: False
 * /oak/camera/i_nn_type: none
 * /oak/camera/i_pipeline_type: RGBD
 * /oak/camera/i_publish_tf_from_calibration: True
 * /oak/camera/i_usb_port_id: 
 * /oak/camera/i_usb_speed: SUPER_PLUS
 * /oak/camera_i_base_frame: oak-d_frame
 * /oak/camera_i_cam_pitch: 0.0
 * /oak/camera_i_cam_pos_x: 0.0
 * /oak/camera_i_cam_pos_y: 0.0
 * /oak/camera_i_cam_pos_z: 0.0
 * /oak/camera_i_cam_roll: 0.0
 * /oak/camera_i_cam_yaw: 0.0
 * /oak/camera_i_imu_from_descr: False
 * /oak/camera_i_parent_frame: oak-d-base-frame
 * /oak/camera_i_publish_tf_from_calibration: False
 * /oak/imu/i_enable_rotation: True
 * /oak/imu/i_message_type: IMU
 * /oak/pipeline_gen/i_enable_imu: True
 * /oak/rgb/i_height: 2160
 * /oak/rgb/i_resolution: 4K
 * /oak/rgb/i_width: 3840
 * /oak/rgb/r_set_man_exposure: True
 * /oak/rgb/r_set_man_whitebalance: True
 * /oak/robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    oak (nodelet/nodelet)
    oak_nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rectify_color (nodelet/nodelet)
    set_log_level (rosservice/rosservice)

ROS_MASTER_URI=http://localhost:11311

process[set_log_level-1]: started with pid [740]
process[oak_state_publisher-2]: started with pid [741]
process[oak_nodelet_manager-3]: started with pid [742]
process[oak-4]: started with pid [743]
process[rectify_color-5]: started with pid [744]
[ INFO] [1732665629.379231039]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1732665629.387315903]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1732665629.397538972]: Initializing nodelet with 8 worker threads.
[ INFO] [1732665629.408534075]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ INFO] [1732665629.584955836]: Starting camera.
[2024-11-27 00:00:29.585] [depthai] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2024-11-27 00:00:29.585] [depthai] [debug] Library information - version: 2.28.0, commit:  from , build: 2024-08-22 15:09:00 +0000, libusb enabled: true
[2024-11-27 00:00:29.588] [depthai] [debug] Initialize - finished
[ INFO] [1732665629.591132297]: No ip/mxid/usb_id specified, connecting to the next available device.
[2024-11-27 00:00:29.683] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.28.tar.xz' open: 3ms, archive read: 94ms
[... 970 lines cut ...]
    r_set_man_exposure: true
```

Oh, and one more thing I’ve noticed: the OAK-D-W model uses the BNO085 IMU, but in the file `depthai-ros/depthai_ros_driver/src/param_handlers/imu_param_handler.cpp`, it seems like it only allows activation if the `IMU` is a `BNO086`. Please check this.

### depthai-ros/depthai_ros_driver/src/param_handlers/imu_param_handler.cpp
```
bool rotationAvailable = imuType == "BNO086";
if(declareAndLogParam<bool>("i_enable_rotation", false)) {
    if(rotationAvailable) {
        auto rotationVecType = utils::getValFromMap(utils::getUpperCaseStr(declareAndLogParam<std::string>("i_rotation_vector_type", "ROTATION_VECTOR")),
                                                    rotationVectorTypeMap);
        imu->enableIMUSensor(rotationVecType, declareAndLogParam<int>("i_rot_freq", 400));
        // if imu message type is IMU_WITH_MAG or IMU_WITH_MAG_SPLIT, enable magnetometer
        if(messageType == "IMU_WITH_MAG" || messageType == "IMU_WITH_MAG_SPLIT") {
            imu->enableIMUSensor(dai::IMUSensor::MAGNETOMETER_CALIBRATED, declareAndLogParam<int>("i_mag_freq", 100));
        }
    } else {
        ROS_ERROR("Rotation enabled but not available with current sensor");
        declareAndLogParam<bool>("i_enable_rotation", false, true);
    }
}
```

Thank you for your help, Serafadam.

dbparkJ, 2024-11-27:
[image]
[image]

I just tested with the Python code available at https://docs.luxonis.com/software/depthai/examples/imu_accelerometer_gyroscope/ and got the following result:
```
jm@jm-desktop:~$ python test.py 
IMU type: BMI270, firmware version: 1.0.0
Rotation vector output is supported only by BNO086!
```
I have the OAK-D-W model, and the sales website (https://shop.luxonis.com/products/oak-d-w?variant=43905772519647) also mentions that it uses the BNO085 sensor. 
[image]
Can you help me understand why I'm seeing this message?

If the sensor is indeed BMI270, then wouldn't that mean I was scammed? Am I right?

Serafadam, 2024-11-27:
Hi, 
1. BNO086 is just BNO085 with updated firmware, so that shouldn't matter in the end
2. From the logs I can see that IMU parameters are not set properly ROS1 and ROS2 differ in parameter naming convention due to different API in ROS libraries, so the second parameter file that you posted is ROS2 based, for ROS1 you follow the convention of the first config you posted. As can be seen when running this param file, logs output
```
 * /oak/imu/i_enable_rotation: True
 * /oak/imu/i_message_type: IMU
 ```
 which should be
 ```
  * /oak/imu_i_enable_rotation: True
 * /oak/imu_i_message_type: IMU
```
for ROS1
3. Some earlier models had BMI270 chip installed due to chip shortage (this was mentioned on the older documentation webpage but seems missing in the new one, we apologize for that), please contact our support about replacement.

dbparkJ, 2024-11-27:
Thank you @Serafadam 
I sent an email to support team, thank you for your help.

closed 2024-11-27 by dbparkJ