luxonis/depthai-ros #638: [BUG] problem synchronizing cameras, with depth and imu

DanielCyberbee, 2024-12-03:
I can't synchronize the cameras, depth and imu with "message_filters::TimeSynchronizer" but I can synchronize them with "message_filters::sync_policies::ApproximateTime". 
It happens because the time stamps are not exactly alien and I do need them to be properly alien. 
I'm using the depthai_examples stereo_node node
Any help?

DanielCyberbee, 2024-12-03:
ti be exact. the right and depth are synchronized but the left and imu are  the problematic ones. here is a image representing it
[image]

Serafadam, 2024-12-04:
Hi, each sensors reports its own timestamp of when the data was captured. In DepthAI C++/Python library we can use `Sync` node to sync between those different sensors although then all data is being output at the frequency of the lowest sensor. For image publishers this mechanism is actually implemented in `depthai_ros_driver`, you can check out documentation on how it works, it's just a matter of enabling few parameters in a config file. To enable that in stereo_node you would need to edit the source code to add that functionality, although that might be a bit more complex. When it comes to IMU data, the resulting plot looks that way due to how IMU driver outputs the data from the device, this might change when we release V3 version of DepthAI in the future

DanielCyberbee, 2024-12-04:
Is there a simple way to synchronize the stereo cameras? Maybe even a hard code approach to go along the stereo_node.
If so, what do I need to change?

DanielCyberbee, 2024-12-08:
what functionalities do i need to add to the stereo_node? 
I'm having a hard time implementing it.

Serafadam, 2024-12-09:
Hi, you can refer to exampled in [depthai-core](https://github.com/luxonis/depthai-core/blob/main/examples/Sync/depth_video_synced.cpp) on pipeline setup, with a sync node it's unfortunately more complicated when it comes to DAI_ROS bridge as is since it's designed for single input queues. In short, you would need to create a separate ROS node with publishers and for each message publish that message, similarly to how it's done in [depthai_ros_driver](https://github.com/luxonis/depthai-ros/blob/da0463f2126998bae6ce1629b09d326e97ae20b3/depthai_ros_driver/src/dai_nodes/sensors/sync.cpp#L37).