luxonis/depthai-ros #641: [Issue] Drop publish rate of depth pointcloud

aminyaa, 2024-12-09:
I'm using depthai-ros-driver (ros2 humble) and my device is OAK-D-LITE. The problem is, when I run the poitcloud launch file, everything is OK and the publish rate of the pointcloud is about 30 hz. But after a while (almost 20 minuets), the publish rate drops (to almost 1 hz). The camera heats up (I think it is normal), but when I reset the node (rerun the launch poincloud file), everything will be OK and the publish rate is 30 hz, but it drops again after 20 minutes to 1 hz again.

Serafadam, 2024-12-09:
Hi, this could be related to your network/DDS implementation, please try changing to another DDS or [tune the current one](https://docs.ros.org/en/jazzy/How-To-Guides/DDS-tuning.html) and see if the issues persists.

advaitp, 2024-12-18:
Hi I am still facing this issue even when I use Best Effort QoS settings.I am using OAK D Pro W PoE camera and I am getting depth map at around 9 to 10 Hz but when I check the pointcloud topic it falls after around 15 minute to less than 1Hz.I am running "ros2 launch depthai_examples stereo.launch.py" from depthai-ros repository.

Serafadam, 2024-12-19:
Hi, have you tried using a different DDS implementation?

Serafadam, 2025-01-07:
Hi, it all depends on you setup and needs, but usually people report improvements when switching to CycloneDDS

TanayK07, 2026-10-18:
one thing that may help narrow it: the cloud isn't published by the camera driver itself. both pointcloud launch files build it in depth_image_proc's PointCloudXyziNode inside the camera container, and that node pairs the depth image with the right rectified image and camera_info by timestamp. so there are two different failures that look the same from outside: that node only producing ~1 hz (inputs stopped pairing up, or it's falling behind), or it still producing 30 and the big clouds getting lost on the way to your subscriber, which is where the dds advice applies. when it's in the slow state, ros2 topic hz on the camera_info topics is cheap and tells you if the inputs are still at full rate. to get the pointcloud node's actual publish rate from inside the container, i wrote ros2_pulse: LD_PRELOAD it into the launch and it logs publish hz and per-input receive hz per topic for the whole run, intra-process included, without adding a subscriber on the cloud (it counts only, so stamp drift between inputs you'd still check by hand). https://github.com/TanayK07/ros2_pulse