luxonis/depthai-ros #652: How can I reduce point cloud size with ros driver?

bartonp2, 2025-01-08:
I have an OAK-D Pro PoE Fixed Focus camera which I am launch with the ros driver. I only need the pointcloud with 3Hz and would like to reduce the resolution of the point cloud as well, such that I don't have to do it within software. I am able to set the resolution of the rgb and left image, however, upon changing the width and height of the right or stereo image, I get the following error:
```
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzi' in container 'oak_container'               
[component_container-1] terminate called after throwing an instance of 'cv::Exception'                                        
[component_container-1]   what():  OpenCV(4.2.0) /home/ubuntu/build_opencv/opencv/modules/core/src/matrix.cpp:768: error: (-21
5:Assertion failed) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function 'Mat'     
[component_container-1]
```

Here is the config I am using:
```
/oak:
  ros__parameters:
    rgb:
      i_publish_topic: false
      i_fps: 3.0
    left:
      i_fps: 3.0                  # Also defines the point cloud fps - 2fps leads to auto exposure problems, other fps seem to work
      i_width: 640              
      i_height: 400
    right:
      i_fps: 3.0
      # i_width: 640              # cannot seem to set right to 400p. Works on left
      # i_height: 400
    stereo:
      i_publish_topic: true       # point cloud not published without this
      i_align_depth: true
      i_board_socket_id: 2
      i_subpixel: true
      i_publish_right_rect: true  # needed for point cloud to be published
      i_right_rect_synced: false
      # i_width: 640              
      # i_height: 400
    camera:
      i_nn_type: none
      i_enable_imu: false
```

What is the recommended way to change the resolution of the pointcloud and which other settings are required for it to work?

Serafadam, 2025-01-10:
Hi, to properly set width/height you should start with i_resolution parameter (set left/right.i_resolution: 400P, width and height should be updated automatically)

gurbain, 2025-01-10:
Hello,

I am facing the same problem here with an OAK-D. Is there an example somewhere with reduced size pointcloud? I think that it is quite useful in some robotics application. With these parameters:

```
/oak:
  ros__parameters:
    camera:
      i_pipeline_type: RGBD
    stereo:
      i_publish_topic: true 
      enable_spatial_nn: true
    right:
      i_resolution: 400P
      i_fps: 5.0
      i_low_bandwidth: true
    left:
      i_resolution: 400P
      i_fps: 5.0
      i_low_bandwidth: true
```
The point cloud is published in 1280x720p. And If I change it to:
```
/oak:
  ros__parameters:
    camera:
      i_pipeline_type: RGBD
    stereo:
      i_publish_topic: true 
      i_height: 400
      i_width: 640
    right:
      i_resolution: 400P
      i_fps: 5.0
      i_low_bandwidth: true
    left:
      i_resolution: 400P
      i_fps: 5.0
      i_low_bandwidth: true
```
I get the error:
```
 what():  OpenCV(4.5.4) ./modules/core/src/matrix.cpp:768: error: (-215:Assertion failed) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function 'Mat'
```

Any idea? thank you

bartonp2, 2025-01-10:
Great, thanks this worked @Serafadam :+1: 

However, I noticed that having installed the depthai-ros-driver via apt, I do not have the latest code. Maybe this is causing some problems for @gurbain .
I am on Ubuntu focal 20.04, aarch64-linux-gnu and installed ros-humble-depthai*

bartonp2, 2025-01-13:
@Serafadam When can we expect such a sync or is there some way to track this?

Serafadam, 2025-01-13:
I would expect that the packages will be released within the week as last one was around month ago, you can check for updates on ROS forums https://discourse.ros.org/tag/release
You can also use testing repository to get the latest binaries https://docs.ros.org/en/jazzy/Installation/Testing.html