luxonis/depthai-ros #654: [BUG] Build Errors

dbparkJ, 2025-01-13:
## Hardware and OS Information
- **Hardware**: Jetson Orin NX 16GB
- **OS**: JetPack 5.1.4
- **ROS Version**: Noetic (desktop-full)

---
## Issue Description
While attempting to build `depthai_ros_driver` using `catkin_make`, the build process fails with the following errors:

```
[ 87%] Building CXX object depthai-ros/depthai_ros_driver/CMakeFiles/depthai_ros_driver_common.dir/src/param_handlers/sync_param_handler.cpp.o
/home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp: In constructor ‘depthai_ros_driver::param_handlers::StereoParamHandler::StereoParamHandler(ros::NodeHandle, const string&)’:
/home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp:13:71: error: ‘DEFAULT’ is not a member of ‘dai::node::StereoDepth::PresetMode’
   13 |                       {"DEFAULT", dai::node::StereoDepth::PresetMode::DEFAULT},
      |                                                                       ^~~~~~~
/home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp:14:68: error: ‘FACE’ is not a member of ‘dai::node::StereoDepth::PresetMode’
   14 |                       {"FACE", dai::node::StereoDepth::PresetMode::FACE},
      |                                                                    ^~~~
/home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp:15:75: error: ‘HIGH_DETAIL’ is not a member of ‘dai::node::StereoDepth::PresetMode’; did you mean ‘HIGH_DENSITY’?
   15 |                       {"HIGH_DETAIL", dai::node::StereoDepth::PresetMode::HIGH_DETAIL},
      |                                                                           ^~~~~~~~~~~
      |                                                                           HIGH_DENSITY
/home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp:16:72: error: ‘ROBOTICS’ is not a member of ‘dai::node::StereoDepth::PresetMode’
   16 |                       {"ROBOTICS", dai::node::StereoDepth::PresetMode::ROBOTICS}};
      |                                                                        ^~~~~~~~
/home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp:16:81: error: no match for ‘operator=’ (operand types are ‘std::unordered_map<std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode>’ and ‘<brace-enclosed initializer list>’)
   16 |                       {"ROBOTICS", dai::node::StereoDepth::PresetMode::ROBOTICS}};
      |                                                                                 ^
In file included from /usr/include/c++/9/unordered_map:47,
                 from /home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/include/depthai_ros_driver/param_handlers/stereo_param_handler.hpp:4,
                 from /home/jm/workspace/utils/dai_ws/src/depthai-ros/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp:1:
/usr/include/c++/9/bits/unordered_map.h:272:7: note: candidate: ‘std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>& std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::operator=(const std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>&) [with _Key = std::__cxx11::basic_string<char>; _Tp = dai::node::StereoDepth::PresetMode; _Hash = std::hash<std::__cxx11::basic_string<char> >; _Pred = std::equal_to<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode> >]’
  272 |       operator=(const unordered_map&) = default;
      |       ^~~~~~~~
/usr/include/c++/9/bits/unordered_map.h:272:17: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const std::unordered_map<std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode>&’
  272 |       operator=(const unordered_map&) = default;
      |                 ^~~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/unordered_map.h:276:7: note: candidate: ‘std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>& std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::operator=(std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>&&) [with _Key = std::__cxx11::basic_string<char>; _Tp = dai::node::StereoDepth::PresetMode; _Hash = std::hash<std::__cxx11::basic_string<char> >; _Pred = std::equal_to<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode> >]’
  276 |       operator=(unordered_map&&) = default;
      |       ^~~~~~~~
/usr/include/c++/9/bits/unordered_map.h:276:17: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘std::unordered_map<std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode>&&’
  276 |       operator=(unordered_map&&) = default;
      |                 ^~~~~~~~~~~~~~~
/usr/include/c++/9/bits/unordered_map.h:290:7: note: candidate: ‘std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>& std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::operator=(std::initializer_list<typename std::_Hashtable<_Key, std::pair<const _Key, _Tp>, _Alloc, std::__detail::_Select1st, _Pred, _Hash, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<std::__not_<std::__and_<std::__is_fast_hash<_Hash>, std::__is_nothrow_invocable<const _Hash&, const _Key&> > >::value, false, true> >::value_type>) [with _Key = std::__cxx11::basic_string<char>; _Tp = dai::node::StereoDepth::PresetMode; _Hash = std::hash<std::__cxx11::basic_string<char> >; _Pred = std::equal_to<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode> >; typename std::_Hashtable<_Key, std::pair<const _Key, _Tp>, _Alloc, std::__detail::_Select1st, _Pred, _Hash, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<std::__not_<std::__and_<std::__is_fast_hash<_Hash>, std::__is_nothrow_invocable<const _Hash&, const _Key&> > >::value, false, true> >::value_type = std::pair<const std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode>]’
  290 |       operator=(initializer_list<value_type> __l)
      |       ^~~~~~~~
/usr/include/c++/9/bits/unordered_map.h:290:46: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘std::initializer_list<std::pair<const std::__cxx11::basic_string<char>, dai::node::StereoDepth::PresetMode> >’
  290 |       operator=(initializer_list<value_type> __l)
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
make[2]: *** [depthai-ros/depthai_ros_driver/CMakeFiles/depthai_ros_driver_common.dir/build.make:193: depthai-ros/depthai_ros_driver/CMakeFiles/depthai_ros_driver_common.dir/src/param_handlers/stereo_param_handler.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4616: depthai-ros/depthai_ros_driver/CMakeFiles/depthai_ros_driver_common.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed
```

Thank you!

SeokHyunAhn, 2025-01-13:
@Serafadam I had the same issue with ROS2 humble.

Serafadam, 2025-01-13:
Hi, newest Humble branch is tied to DepthAI 2.29, both versions should be available with the next ROS sync, until then you can either use previous commit from humble branch or use ROS2 testing repositrory to get newest binaries https://docs.ros.org/en/jazzy/Installation/Testing.html until they are released