luxonis/depthai-ros #656: [Feature-Request] {OAK-Thermal ROS Noetic support}

pala5, 2025-01-16:
Hi!

First, I know that there is available the [humble_thermal](https://github.com/luxonis/depthai-ros/tree/humble_thermal) branch, but I am wondering if you will release the ROS Noetic compatible branch. 
I am aware of the close EoL of the ROS Noetic distro, but I am currently working in a project fully implemented in ROS1 Noetic, and at this point it is not possible to migrate the solution.

Thanks!

Serafadam, 2025-01-16:
Hi, you can expect noetic equivalent branch in few days, note that this api isn't final yet (and Thermal camera itself doesn't have full API exposed in current firmware, this will be available in V3 release which is going to be available sometime later this year)

pala5, 2025-01-17:
Hi @Serafadam!

Thank you for your quick response. 
Where can I find the limitations of the current firmware of the OAK-T?

Thanks!

Serafadam, 2025-01-17:
Hi, in current V2 of firmware thermal camera is implemented by generic camera node as can be seen in [this example](https://github.com/luxonis/depthai-core/blob/main/examples/Camera/thermal_cam.cpp).  In V3 there is a [dedicated thermal camera node ](https://github.com/luxonis/depthai-core/blob/v3_develop/examples/cpp/RVC2/Thermal/thermal.cpp)

Serafadam, 2025-01-20:
Hi @pala5 you can find initial port for Noetic on [this branch](https://github.com/luxonis/depthai-ros/tree/noetic_thermal), similarly to humble you can use "camarray" pipeline type to use the camera for now

pala5, 2025-01-21:
Hi @Serafadam, thanks a lot! I will try it ASAP.

closed 2025-01-21 by pala5