luxonis/depthai-ros #661: [Feature-Request] {Running OAK-T in humble_thermal branch}

DoppiaEffe94, 2025-02-10:
Hello!

I'm trying to configure my OAK-T with ROS 2.

I know there is a branch called **humble_thermal**, however the launch file does not seems to be properly configured for launching the nodes related to the OAK-T.

For example, launching the **depthai_ros_driver/camera.launch.py** is throwing me these errors:

`component_container-1] "[1739180537.819806723 oak 81] Starting camera." oak 81] Starting camera.}
[component_container-1] "[1739180537.822321197 oak 201] No ip/mxid specified, connecting to the next available device." oak 201] No ip/mxid specified, connecting to the next available device.}
[component_container-1] "[1739180545.186915088 oak 255] Camera with MXID: 14442C10F1C2A2D200 and Name: 169.254.1.222 connected!" oak 255] Camera with MXID: 14442C10F1C2A2D200 and Name: 169.254.1.222 connected!}
[component_container-1] "[1739180545.186973527 oak 262] PoE camera detected. Consider enabling low bandwidth for specific image topics (see Readme->DepthAI ROS Driver->Specific camera configurations)." oak 262] PoE camera detected. Consider enabling low bandwidth for specific image topics (see Readme->DepthAI ROS Driver->Specific camera configurations).}
[component_container-1] "[1739180545.234464548 oak 160] Device type: OAK-T" oak 160] Device type: OAK-T}
[component_container-1] "[1739180545.238902344 oak 48] Pipeline type: RGBD" oak 48] Pipeline type: RGBD}
[component_container-1] "[1739180545.288931690 oak 108] Invalid pipeline chosen for camera as it has only stereo pair. Switching to Depth." oak 108] Invalid pipeline chosen for camera as it has only stereo pair. Switching to Depth.}
[component_container-1] terminate called after throwing an instance of 'std::out_of_range'
[component_container-1]   what():  _Map_base::at
[ERROR] [component_container-1]: process has died [pid 1845, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level info --ros-args -r __node:=oak_container -r __ns:=/'].`

Am I doing something wrong?

**I'm just wondering if there could be a possibility to have more info on how to launch the camera properly.**

Thank you!

pala5, 2025-02-11:
Hi,

I take advantage of the open issue to say that there is also this problem in the branch [noetic_thermal](https://github.com/luxonis/depthai-ros/tree/noetic_thermal). 
I have done some research, and it seems that there are missing the OAK-T config and plugins.xml files.

Thanks!

Serafadam, 2025-02-11:
Hi, for launching OAK-T on these branches we suggest using CamArray pipeline type, a specific thermal pipeline will be added soon but will mostly work in the same way

Serafadam, 2025-02-11:
Thermal pipeline/launch file[has been added to the branch](https://github.com/luxonis/depthai-ros/commit/9690e70cb39257f5cada66877c35726cc761e225), changes will be merged to Humble soon

pala5, 2025-02-12:
Hi @Serafadam, are you considering making the same modifications for the _noetic_thermal_ branch? 
Thanks!

Serafadam, 2025-02-12:
Hi, yes, PR should be up today

closed 2025-02-18 by DoppiaEffe94