luxonis/depthai-ros #664: [BUG] depthai-ros ROS1 metapackage not properly defined in package.xml

xaru8145, 2025-02-14:
**Describe the bug**
The depthai-ros metapackage should include <metapackage /> export tag in ROS1 (noetic) so that catkin tools treats it as metapackage rather than package, as stated in the [wiki](https://wiki.ros.org/catkin/package.xml#Metapackages).

**Minimal Reproducible Example**
Dry-run compilation of depthai-ros package `catkin build --dry-run depthai-ros` outputs the following:

```
Packages to be built:
- depthai-ros        (catkin)
- depthai_ros_msgs   (catkin)
- depthai_bridge     (catkin)
- xacro              (catkin)
- depthai_examples   (catkin)
- depthai_ros_driver (catkin)
Total packages: 6
```

**Expected behavior**
Dry-run compilation of depthai-ros package `catkin build --dry-run depthai-ros` **should** output the following:

```
Packages to be built:
- depthai-ros        (metapackage)
- depthai_ros_msgs   (catkin)
- depthai_bridge     (catkin)
- xacro              (catkin)
- depthai_examples   (catkin)
- depthai_ros_driver (catkin)
Total packages: 6
```

Serafadam, 2025-02-14:
Hi, does this affect your build in significant way?

xaru8145, 2025-02-14:
Yes, I am trying to automate deb package generation for the packages in a workspace. For that, I run catkin build --dry-run and get the packages (excluding metapackages) in order based on dependencies. If catkin sees depthai-ros as a package, it will try to generate the deb package and fail since its real package dependencies will be listed after. I am open to suggestions to solve it differently.

Thanks for the quick response :)

closed 2025-03-07 by xaru8145