luxonis/depthai-ros #666: [BUG] Incorrect IMU mapping in rtabmap launch file

rdlynx19, 2025-02-16:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
I am trying out the example rtabmap launch file by running `ros2 launch depthai_ros_driver rtabmap.launch.py`. However, as soon as the rtabmap-viz opens, it shows odometry lost.

**Expected behavior**
rtabmap-viz should open up correctly and the mapping should begin.

**Screenshots**
[image]

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
 - OS: Ubuntu 24.04
 - ROS2 Jazzy
```
 - redhairedlynx@tokyo:~$ apt show ros-jazzy-depthai
Package: ros-jazzy-depthai
Version: 2.29.0-1noble.20250102.150449
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 68.2 MB
Depends: libc6 (>= 2.38), libgcc-s1 (>= 4.2), libopencv-core406t64 (>= 4.6.0+dfsg), libopencv-imgproc406t64 (>= 4.6.0+dfsg), libstdc++6 (>= 13.1), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-jazzy-ros-workspace
Homepage: https://www.luxonis.com/
Download-Size: 24.9 MB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu noble/main amd64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform
```
```
redhairedlynx@tokyo:~$ apt show ros-jazzy-depthai-ros
Package: ros-jazzy-depthai-ros
Version: 2.10.5-1noble.20250115.212226
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 44.0 kB
Depends: ros-jazzy-depthai, ros-jazzy-depthai-bridge, ros-jazzy-depthai-descriptions, ros-jazzy-depthai-examples, ros-jazzy-depthai-filters, ros-jazzy-depthai-ros-driver, ros-jazzy-depthai-ros-msgs, ros-jazzy-ros-workspace
Download-Size: 7,050 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu noble/main amd64 Packages
Description: The depthai-ros package
```

**Additional context**
I was able to fix this bug for my system by remapping the imu topic that rtabmap listens to from `/imu` to `oak/imu/data`
However, even after that I observe the point cloud is bouncing around. Important thing to note here is the camera is completely stable and not moving, then what is causing this bouncing around?

https://github.com/user-attachments/assets/f8791ae1-5c39-47ab-b45c-8ffd5986145f

Serafadam, 2025-02-17:
Hi, thanks for reporting, actually the first error that you receive (mapping not starting right up) can be a result of autofocus/autoexposure values, you can either set those values or trigger an odometry reset inside RTABMap. As for the rotation part, I think there might be some tuning needed in RTABMap itself, we don't pass IMU data in the example since default RTABMap's VO algorithms don't utilize (they only operate on visual data, IMU is just for initial orientation guess and that is usually handled by setting up correct transforms on a robot)

rdlynx19, 2025-02-19:
Hi @Serafadam! Thanks for the reply.
I'm actually using the `depthai.launch.py` launch file from the rtabmap examples. I was using this launch file as a way to debug whether the issue was from the camera or rtabmap. 
More details on this issue on rtabmap: https://github.com/introlab/rtabmap_ros/issues/1270
In the `depthai.launch.py` file as well, the imu frame, seems to be rotated with respect to the oak base frame. I found a temporary fix by modifying the urdf of the camera. But I don't think that is the correct way. Do you have any thoughts on this?