luxonis/depthai-ros #675: [BUG] distorted decimated pointcloud

nturro, 2025-02-19:
hello,

when decimating a pointcloud, the pointcloud is distorted.

The correct pointcloud is displayed here on rviz :

[image]

The distorted (same camera angle view) : 

[image]

It might be related to https://github.com/luxonis/depthai-python/issues/1034

environement : luxonis/depthai-ros:humble-latest container 
 
command (pointcloud.launch.py is the original one):
``` 
ros2 launch depthai_ros_driver pointcloud.launch.py params_file:='/root/luxonis/pcl.yaml
```

tweaked camera config pcl.yaml

```
/oak:
  ros__parameters:
    camera:
      i_nn_type: none
    left:
      i_resolution: '400P'
    right:
      i_resolution: '400P'

    stereo:
      i_align_depth: true
      i_board_socket_id: 2
      i_subpixel: true
      i_right_rect_publish_topic: true
      i_right_rect_synced: false
      i_enable_decimation_filter: true
      i_decimation_filter_decimation_factor: 1
      i_decimation_filter_decimation_mode: PIXEL_SKIPPING
```

Could you help me ?

Erol444, 2025-02-19:
@Serafadam , could you please check this? Seems like nturro has latest depthai, but from the images it does seem pointcloud is incorrect. Not sure if this is ROS issue or depthai's..

Serafadam, 2025-02-19:
Hi, so far I am unable to replicate this issue on latest Humble version, you could either wait till new image gets released (probably later today or tomorrow) or build the docker image yourself following instructions [here](https://docs.luxonis.com/software/ros/depthai-ros/build/#Docker-Building)

nturro, 2025-02-20:
hello, i followed the instructions and rebuild the image, and still have the same problem.

As far as i understand, the dockerfile only recompiles the ros wrapper, while the the libdepthai-core.so is the same (coming from the ros-humble 'docker') : 

`
root@ari-10c:/ws# ldd ./build/depthai_ros_driver/libdepthai_ros_driver.so | grep libdepth
	libdepthai_ros_driver_sensor_nodes.so => /ws/install/lib/libdepthai_ros_driver_sensor_nodes.so (0x00007fbbda182000)
	libdepthai_ros_driver_nn_nodes.so => /ws/install/lib/libdepthai_ros_driver_nn_nodes.so (0x00007fbbd9f08000)
	libdepthai_ros_driver_common.so => /ws/install/lib/libdepthai_ros_driver_common.so (0x00007fbbd9a76000)
	libdepthai_bridge.so => /ws/install/lib/libdepthai_bridge.so (0x00007fbbd9807000)
	libdepthai-core.so => /opt/ros/humble/lib/x86_64-linux-gnu/libdepthai-core.so (0x00007fbbd7d07000)
	libdepthai_ros_msgs__rosidl_typesupport_cpp.so => /ws/install/lib/libdepthai_ros_msgs__rosidl_typesupport_cpp.so (0x00007fbbd784d000)
`

Serafadam, 2025-02-20:
Hi, new image is up, you can update it with
`docker pull luxonis/depthai-ros:humble-latest`,
the pointcloud example should run normally, if that's not the case we will investigate further

nturro, 2025-02-20:
Hi,
Thanks for your feedback.

i just pulled and tested the new image, same result : 

[image]

This is using i_decimation_filter_decimation_factor: 4 , the point cloud is even more tilted into the ground.

I also tried to recompile https://github.com/luxonis/depthai-core and use the it in the container to have the latest version, no improvement.

Serafadam, 2025-02-21:
Hi, I think I know what the issue is, since the launch file is using PointCloudxyzi which uses right rectified image for intensity it seems that it expects both right and stereo images to have the same dimensions and due to using decimation filter the final dimension of the images is different.
Unfortunately changing width/height of the right image node won't help here since its connected both to the topic and stereo. I can see two options here:
- Adding an intermediate resize node - unfortunately it seems that the one from `image_proc` package uses different QoS for publishing which prevents pointcloud node from subscribing to it, you would probably need to create a separate package, copy it and change the code to use different QoS for publishing, which is more work
- Update/create new launch file that uses PointCloudXYZ instead, it's just a matter of changing these lines:
```
ComposableNode(
                    package="depth_image_proc",
                    plugin="depth_image_proc::PointCloudXyziNode",
                    name="point_cloud_xyzi",
                    remappings=[
                        ("depth/image_rect", name + "/stereo/image_raw"),
                        ("intensity/image_rect", name + "/right/image_rect"),
                        ("intensity/camera_info", name + "/stereo/camera_info"),
                        ("points", name + "/points"),
                    ],
                ),
```
to 
```
ComposableNode(
                    package="depth_image_proc",
                    plugin="depth_image_proc::PointCloudXyzNode",
                    name="point_cloud_xyz",
                    remappings=[
                        ("image_rect", name + "/stereo/image_raw"),
                        ("camera_info", name + "/stereo/camera_info"),
                        ("points", name + "/points"),
                    ],
                ),
```
Please let me know if that also works for you