luxonis/depthai-ros #681: [BUG] Right camera distortion

xhu4, 2025-02-24:
**Check if issue already exists**

- [x] Google it (e.g. *error xy* github luxonis depthai)
- [x] Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
Artifacts and distortion in raw camera image when using `depthai_ros_driver`, and the distortion is affecting depth image. However there is no issue with `depthai_viewer`. See screenshots below for details.

**Minimal Reproducible Example**
This issue might not be consistently reproducible but it does occur to us multiple times with different oak-d-pro cameras. We first had this issue and thought it was a hardware problem and switched to a different device. The new device worked fine at first but after a while started to have this issue again. I'm hoping that someone might know what could be the issue based on the symptoms we have observed so far.

I used the following command to run the ros driver with rviz.
```
> docker run --rm -it --privileged -v /dev/:/dev/ -v /tmp/.X11-unix:/tmp/.X11-unix --env DISPLAY=$DISPLAY --network=host luxonis/depthai-ros:v2.11.0-humble bash
# ros2 launch depthai_ros_driver pointcloud.launch.py use_rviz:=true
```

Append the MRE to the bug report, [instructions here](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example)

> There was no crash dump found on your device!

_FYI the instruction link is invalid._

**Expected behavior**

There should be no distortion in the right camera. The image should be the same as in the depthai_viewer.

**Screenshots**

This is what it looks like in rviz2, note the circle of dots and distortion near left and right edges:
[image]

This is the right camera from `depthai_viewer`
[image]

This is not an rviz issue since the point cloud itself is also affected by this distortion (see up left corner):
[image]

Comparing to depthai viewer:
[image]

**Pipeline Graph**

NA

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
 - Which OS/OS version are you using?
 - - Ubuntu 22.04

To the following questions: docker image `luxonis/depthai-ros:v2.11.0-humble` is used.
 - Which ROS version are you using?
 - Which ROS distribution are you using ?
 - Is `depthai-ros` built from source or installed from apt?
 - Is `depthai/depthai-core` library installed from rosdep or manually? For rosdep install, check if `ros-<rosdistro>-depthai` package is installed, manual install can be checked with `ldconfig -p | grep depthai`
 - Please include versions of following packages - `apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver`
 

 - To get additional logs, set `DEPTHAI_DEBUG=1` and paste the logs, either from command line or from latest log in `~/.ros/log`
 - - Attached [log.zip](https://github.com/user-attachments/files/18948232/log.zip)

Serafadam, 2025-02-25:
Hi, it seems that there might be an issue with the device calibration, what device type are you using? It seems that depthai-viewer uses RGB sensor for pointcloud creation, could you verify that your pointcloud looks better by using camera.launch.py with `pointcloud.enable:=true`?

xhu4, 2025-02-25:
Hi! Thanks for the response! I'm using an oak-d-pro. I got this error when running camera.launch.py with `pointcloud.enable:=true`:

```
[component_container-2] [INFO] [1740510296.969531011] [oak]: Device type: OAK-D-PRO
[component_container-2] [INFO] [1740510296.975065288] [oak]: Pipeline type: RGBD
[component_container-2] [INFO] [1740510297.428577925] [oak]: NN Family: mobilenet
[component_container-2] [INFO] [1740510297.474470640] [oak]: NN input size: 300 x 300. Resizing input image in case of different dimensions.
[component_container-2] [INFO] [1740510297.732246162] [oak]: Finished setting up pipeline.
[component_container-2] [18443010415B3CF500] [1.11] [1.978] [SpatialDetectionNetwork(7)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
[component_container-2] terminate called after throwing an instance of 'std::system_error'
[component_container-2]   what():  Device already closed or disconnected: Input/output error
[ERROR] [component_container-2]: process has died [pid 439, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level info --ros-args -r __node:=oak_container -r __ns:=/'].
```

Serafadam, 2025-02-26:
Hi, this could be due to error with connection, please check you cables and try again