luxonis/depthai-ros #686: [BUG] camera.launch.py not publishing left/right image topics

annanya-tamu, 2025-03-10:
**Description of the bug**
Hello,

I'm trying to get the left and right image topics on my OAK-D Pro W camera. I think it should run with the default camera.launch.py file since the i_left/right_rect_publish_topic is true like like:

`parameter_overrides["depth"] = {
            "i_left_rect_publish_topic": True,
            "i_right_rect_publish_topic": True,
}`

The output of `ros2 launch depthai_ros_driver camera.launch.py` looks like this:
`[INFO] [launch]: All log files can be found below /root/.ros/log/2025-03-10-17-41-12-723270-system76-pc-2145
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [2160]
[component_container-1] [INFO] [1741628473.057221501] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-1] [INFO] [1741628473.062672604] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1741628473.062747875] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1741628473.067772070] [oak_state_publisher]: got segment oak
[component_container-1] [INFO] [1741628473.067785720] [oak_state_publisher]: got segment oak-d-base-frame
[component_container-1] [INFO] [1741628473.067788141] [oak_state_publisher]: got segment oak_imu_frame
[component_container-1] [INFO] [1741628473.067789946] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-1] [INFO] [1741628473.067791749] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-1] [INFO] [1741628473.067793362] [oak_state_publisher]: got segment oak_model_origin
[component_container-1] [INFO] [1741628473.067795036] [oak_state_publisher]: got segment oak_rgb_camera_frame
[component_container-1] [INFO] [1741628473.067796798] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[component_container-1] [INFO] [1741628473.067798652] [oak_state_publisher]: got segment oak_right_camera_frame
[component_container-1] [INFO] [1741628473.067800246] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-1] [INFO] [1741628473.068521721] [oak_container]: Load Library: /ws/install/lib/libdepthai_ros_driver.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container '/oak_container'
[component_container-1] [INFO] [1741628473.105355706] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1741628473.105412450] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1741628473.108336651] [oak_container]: Load Library: /opt/ros/humble/lib/librectify.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-1] [INFO] [1741628473.110193367] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[component_container-1] [INFO] [1741628473.110209798] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/rectify_color_node' in container '/oak_container'
[component_container-1] [INFO] [1741628474.110455628] [oak]: Starting camera.
[component_container-1] [INFO] [1741628474.116254399] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [INFO] [1741628476.529970616] [oak]: Camera with MXID: 14442C105149D1D200 and Name: 1.4 connected!
[component_container-1] [INFO] [1741628476.530832192] [oak]: USB SPEED: SUPER
[component_container-1] [INFO] [1741628476.577653754] [oak]: Device type: OAK-D-PRO-W
[component_container-1] [INFO] [1741628476.580566515] [oak]: Pipeline type: depth
[component_container-1] [ERROR] [1741628476.945835603] [oak]: rgb parameters not set, defaulting to 1280x720 unless specified otherwise.
[component_container-1] [INFO] [1741628477.168196758] [oak]: Finished setting up pipeline.
[component_container-1] [INFO] [1741628477.491862228] [oak]: Camera ready!`

And the output of `ros2 topic list` is:
`/diagnostics
/joint_states
/oak/imu/data
/oak/rgb/camera_info
/oak/rgb/image_raw
/oak/rgb/image_rect
/oak/rgb/image_rect/compressed
/oak/rgb/image_rect/compressedDepth
/oak/rgb/image_rect/theora
/oak/stereo/camera_info
/oak/stereo/image_raw
/oak/stereo/image_raw/compressed
/oak/stereo/image_raw/compressedDepth
/oak/stereo/image_raw/theora
/parameter_events
/robot_description
/rosout
/tf
/tf_static`

**To reproduce the isse**
Connect a OAK-D Pro W camera to a laptop, launch the devcontainer, and run `ros2 launch depthai_ros_driver camera.launch.py`

**Expected behavior**
For the left and right image topics to launch.

I have attached the pipeline graph to this report.

[14442C105149D1D200_pipeline.json](https://github.com/user-attachments/files/19168453/14442C105149D1D200_pipeline.json)

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
 - OS: Ubuntu 22.04
 - ROS: ROS2 Humble
 - `depthai-ros` built from source
 - `depthai/depthai-core` library installed from rosdep
 - The versions of the requested packages are attached to this message.
[package_versions.txt](https://github.com/user-attachments/files/19168493/package_versions.txt)

Serafadam, 2025-03-11:
Hi, those parameters are set if you use RS compatibility mode, so by default driver doesn't publish those topics.

annanya-tamu, 2025-03-11:
Hi @Serafadam, thanks for the quick response!

Is the RS compatibility mode set by the `rs_compact` parameter in the launch file? If so, isn't it already set to false by default here `rs_compat = LaunchConfiguration("rs_compat", default="false")`?

I tried manually setting it when launching by doing `ros2 launch depthai_ros_driver camera.launch.py rs_compat:=false`, but no luck.

Serafadam, 2025-03-12:
Hi, just enabling parameters in camera.yaml (or passing new parameter file with those values) would suffice
```
left:
  i_publish_topic: true
right:
  i_publish_topic: true
```
You can refer [to the documentation](https://docs.luxonis.com/software/ros/depthai-ros/driver/#DepthAI%20ROS%20Driver-List%20of%20parameters) on the list of available parameters

annanya-tamu, 2025-03-12:
Thanks @Serafadam! Changing the `pcl.yaml` file to this
```
/oak:
  ros__parameters:
    camera:
      i_nn_type: none
    stereo:
      i_align_depth: true
      i_board_socket_id: 2
      i_subpixel: true
      i_right_rect_publish_topic: true
      i_right_rect_synced: false
    left:
      i_publish_topic: true
    right:
      i_publish_topic: true
```
and then closing my vscode (where the devcontainer was running), followed by running `docker system prune -a`, then rebuilding the devcontainer fixed my issue. For some reason only the right image topics would publish if I didn't reset everything.