luxonis/depthai-ros #687: [BUG] Image is wrongly formed: step < width * byte_depth * num_channels  or  2560 != 1280 * 1 * 3

BryanBetancur, 2025-03-10:
**Describe the bug**
Hello, I'm testing a OAK-D-PRO-W-POE in order to check if is worth it to change our Realsense robots cameras to OAK cameras but I'm getting this error when I set `i_low_bandwidth: true`, if it's in `false` the camera works properly but the fps are so low.

```
[oak_container-1] [INFO] [1741632856.966561767] [oak]: Starting camera.
[oak_container-1] [INFO] [1741632856.981630463] [oak]: No ip/mxid specified, connecting to the next available device.
[oak_container-1] [INFO] [1741632867.529489182] [oak]: Camera with MXID: 14442C103110F4D600 and Name: 172.16.40.120 connected!
[INFO] [1741632867.529634144] [oak]: PoE camera detected. Consider enabling low bandwidth for specific image topics (see Readme->DepthAI ROS Driver->Specific camera configurations).
[oak_container-1] [INFO] [1741632867.592237216] [oak]: Device type: OAK-D-PRO-W-POE
[oak_container-1] [INFO] [1741632867.623092985] [oak]: Pipeline type: RGBD
[oak_container-1] [INFO] [1741632868.984961750] [oak]: Finished setting up pipeline.
[oak_container-1] [INFO] [1741632869.399505025] [FFMPEGEncoder]: enc: libx264 prof:  preset: 
[INFO] [1741632869.399587362] [FFMPEGEncoder]: qmax: 10 bitrate: 8242880 gop: 15
[oak_container-1] [INFO] [1741632869.796394895] [FFMPEGEncoder]: enc: libx264 prof:  preset: 
[INFO] [1741632869.796475057] [FFMPEGEncoder]: qmax: 10 bitrate: 8242880 gop: 15
[oak_container-1] [INFO] [1741632871.160678255] [oak]: Camera ready!
[oak_container-1] terminate called after throwing an instance of 'cv_bridge::Exception'
  what():  Image is wrongly formed: step < width * byte_depth * num_channels  or  2560 != 1280 * 1 * 3
```

**Minimal Reproducible Example**
I'm using the example launch `camera.launch.py` with the following params file:
```
/oak:
  ros__parameters:
    camera:
      i_nn_type: none
    rgb:
      i_low_bandwidth: true
 
```

**Expected behavior**
After run `ros2 launch depthai_ros_driver camera.launch.py` it will log:
`what():  Image is wrongly formed: step < width * byte_depth * num_channels  or  2560 != 1280 * 1 * 3`.

**Pipeline Graph**

[14442C103110F4D600_pipeline.json](https://github.com/user-attachments/files/19169437/14442C103110F4D600_pipeline.json)
 

**System specifications**
- Jetson Orin AGX 64GB
- OAK-D-PRO-W-POE
- depthai-ros: v2.11.0-humble

Serafadam, 2025-03-12:
Thank you for the report, investigating this now.

Serafadam, 2025-03-12:
Fix: https://github.com/luxonis/depthai-ros/pull/689

HJGrant, 2025-03-13:
I have a similar problem, when I set `i_low_bandwith: true`, I get the following error when I run camera.launch.py:

`[depthai] [error] Callback with id: 1 throwed an exception: Converted type not supported!`

If I set `i_low_bandwith: false`, I don't get the error but the fps are really low, lik 5 fps, when I should be expecting 30 fps.

when I set `i_low_bandwith: true` for the stereo node, I get the following warning:

'[VideoEncoder(13)] [warning] Arrived frame type (14) is not either NV12 or YUV400p (8-bit Gray)'

and nothing is published to the `/oak/stereo/raw_image` and `/oak/stereo/camera_info` topics. Setting i_low_bandwith: false for both nodes solves these issues, but the FPS are very slow. 

I am using the OAK-D PoE.

Serafadam, 2025-03-13:
@HJGrant do you also get the same results when running from [this branch](https://github.com/luxonis/depthai-ros/tree/humble_color_order)?

HJGrant, 2025-03-13:
Unfortunately I get the same error:
```

git branch
  humble
* humble_color_order
```

install/my_oakd_launch/share/my_oakd_launch/config/rgbd.yaml

```
    rgb:
      i_publish_topic: true
      i_resolution: 1080P
      i_fps: 25.0
      i_low_bandwidth: true
      i_output_isp: true
      i_isp_den: 3
      i_isp_num: 2
    stereo:
      i_publish_topic: true
      i_low_bandwidth: true
      i_align_depth: true
      i_lr_check: true
      i_subpixel: true
```

leads to:

```
[depthai] [error] Callback with id: 1 throwed an exception: Converted type not supported!
[VideoEncoder(14)] [warning] Arrived frame type (14) is not either NV12 or YUV400p (8-bit Gray)
```

Serafadam, 2025-03-13:
Hi, unfortunately `subpixel` mode (which produces 16Bit RAW images (type 14)) doesn't currently work with low_bandwidth mode as VideoEncoder that we use in the firmware doesn't support it. Though it should be turned off automatically when enabling low_bandwidth mode. In the logs you should see something like
`[component_container-1] [INFO] [1741864190.175553025] [oak]: Subpixel disabled due to low bandwidth mode`
Did you make changes in the source code? Could you clean up and rebuild your workspace so that we are sure older version of code isn't used?

HJGrant, 2025-03-13:
After cleaning and rebuilding my workspace I can confirm that this branch fixes the problem, thank you!

dwffls, 2025-08-07:
I got the same problem here, however it is when using the `i_publish_left_rect` or `i_publish_right_rect` parameter. 
With this the step appears to be set to 0, and thus the error becomes: `step < width * byte_depth * num_channels or 0 != 1280 * 1 * 1`

When publishing a left image with `left.i_publish_topic` set to true the step is correctly set to 1280;

I've been looking in the code on where this step is set but can't find it anywhere. @Serafadam could you help me out here?