luxonis/depthai-ros #688: [BUG] Joint state not found in URDF due to missing remap in urdf launch

xaru8145, 2025-03-11:
**Describe the bug**
There is a missing remap in the joint_states topic of robot_state_publisher node in [urdf.launch](https://github.com/luxonis/depthai-ros/blob/noetic/depthai_descriptions/launch/urdf.launch). I am using the noetic branch. 

The problem arises when there is another robot_state_publisher publishing robot joints and both nodes are reading and publishing to the same topic, causing a continuous stream of warnings (see screenshot).

In my case, the depthai robot state publisher complains about a motor joint that is in another urd.f The error goes away adding the following joint remap:

```
<node name="$(arg tf_prefix)_state_publisher" pkg="robot_state_publisher"
		type="robot_state_publisher" output="screen" required="true">
        <remap from="robot_description" to="$(arg tf_prefix)/robot_description" />
        <remap from="joint_states" to="$(arg tf_prefix)/joint_states"/>
</node>
```

**Expected behavior**
No warnings should appear.

**Screenshots**

[image]

Serafadam, 2025-03-12:
Hi, I added this change as part of https://github.com/luxonis/depthai-ros/pull/691, you can test that branch and see if it works for you

xaru8145, 2025-03-12:
Yes, works perfectly!Thanks so much

closed 2025-03-12 by xaru8145