luxonis/depthai-ros #698: [BUG] PoE camera not being detected by camera.launch.py

annanya-tamu, 2025-03-21:
**Describe the bug**
Getting the following error within devcontainer when launching `ros2 launch depthai_ros_driver camera.launch.py`:
```bash
[component_container-1] [ERROR] [1742591681.130576375] [oak]: Cannot find any device with given deviceInfo
[component_container-1] [INFO] [1742591681.130666371] [oak]: No ip/mxid specified, connecting to the next available device.
```
I'm confused because DepthAI Viewer is able to detect the camera outside of the container when I run:
```bash
python3 -m depthai_viewer
```
**Minimal Reproducible Example**
Setup: Router -> PoE switch -> OAK-D Pro PoE camera and Laptop. Run the devcontainer given in this repo. Then run `ros2 launch depthai_ros_driver camera.launch.py`.

**Expected behavior**
`camera.launch.py` brings up the PoE camera along with the ROS topics.

**Pipeline Graph**
Services are not available because the launch file is failing.

**Attach system log**
 - Ubuntu 22.04
 - ROS2
 - Humble
 - Built from source

**Additional context**
The router automatically assigns an IP of 192.168.1.6, which the DepthAI Viewer is able to detect. Also I have not seen this error happen with my USB OAK camera.

Serafadam, 2025-03-24:
Hi, are you running depthai-viewer in container as well? Can you ping the camera inside the container?

annanya-tamu, 2025-03-24:
Hi @Serafadam,

When I run `depthai-viewer` inside the container, it does not bring up the camera. I see an empty dashboard:

[image]

However, I am able to ping the camera in the container!

Serafadam, 2025-03-25:
Hi, thanks for that information. I suspect the issue might be related to your Docker setup, although this might be harder to diagnose. Have you tried setting camera IP explicitly via ROS parameter?

annanya-tamu, 2025-03-25:
Hi @Serafadam.

Yes looks like it was an error in my Docker setup. Apologies but I forgot that I had changed my network to be on the `bridge` network.  Changing back to the original in `.devcontainer/devcontainer.json` worked:

```dockerfile
"--network=host",
```

closed 2025-03-25 by annanya-tamu