luxonis/depthai-ros #700: [Feature-Request] Missing OAK-D-W STL file

stanislawix, 2025-03-30:
### Start with the `why`:
Obviously ability to use OAK-D-W with ROS2 without additional hassle.

### Move to the `what`:
STL file for the camera and additional needed files if they are missing in comparison to the other cameras

### Move to the `how`:
You know how guys

Serafadam, 2025-04-01:
Hi, OAK-D-W's model is actually the same as for other S2 variants (like OAK-D Pro/ Pro W etc.) so you can use the default option

stanislawix, 2025-04-03:
Hi, thank you for the information.

Anyways the properly named STL file should still be added to the depthai_descriptions package, so that such error can be omitted: Unable to open file "package://depthai_descriptions/urdf/models/OAK-D-W.stl".

I wonder why it's not already added, especially if its essentially the same as already existing ones.

Serafadam, 2025-04-04:
Hi, introducing additional models that are the same will increase overall package size, changing camera_model parameter is really only useful in cases where the model really differs as OAK-LR or OAK-SR. If you use functionality that publishes TF from calibration directly the model will automatically get selected.
For ROS2 Kilted we will be actually switching to this behavior by default so there won't be need to specify the camera model.