luxonis/depthai-ros #703: oak-1-w-poe image quality poor and low FPS

khuechuong, 2025-04-07:
Okay I have an oak-1-w-poe and to be honest, I'm not sure what's going on. 
Here's what I set my parameters:
/oak:
  camera_i_enable_imu: false
  camera_i_enable_ir: false
  camera_i_floodlight_brightness: 0.0
  camera_i_ip: ''
  camera_i_laser_dot_brightness: 800.0
  camera_i_rgb_i_low_bandwidth: false
  camera_i_fps: 40
  camera_i_mx_id: '14442C10E16A1BD000'
  camera_i_pipeline_type: rgb
  camera_i_usb_port_id: ''
  rgb_i_set_isp_scale: false
  rgb_i_resolution: 2MP

**Expectation:** 
Since I set it only at 2MP and 40FPS, which is lower than what they said on the website since you guys said it could go 12MP with 60FPS. 
**Reality:**
FPS is super low
[image]
Quality is pretty poor:
[image]
[image]

Serafadam, 2025-04-08:
Hi, this looks like a PoE bandwidth issue, you can check our documentation for more details https://docs.luxonis.com/software/depthai/optimizing

khuechuong, 2025-04-08:
@Serafadam So is it a cable issue then? According to that document, 1080P should result in 25 FPS, not 4FPS. My port has capacity of 2.5Gbps. And if the oak-1-w-poe can send 1Gbps, then it should be cable no?

Serafadam, 2025-04-09:
It could be either the cable or network card configuration, you can read up on network testing here https://docs.luxonis.com/hardware/platform/deploy/poe-deployment-guide#PoE%20deployment%20guide-Runtime-Debugging