luxonis/depthai-ros #706: [custom] Camera intrinsic and distortion coeff

khuechuong, 2025-04-08:
I have an OAK-1-W-POE. Now there are different resolution 1080P and others. I was wondering where in the code it publish the camera info for different types like this.

I see a file at ```depthai_ros_driver/config/calibration/rgb.yaml```, but that's for ```image_width: 1280``` and ```image_height: 720```. If I want to run it myself, where do I modify the intrinsic and distortion coeff?  

How accurately is the default? Is it recommended to do my own calibration?

Serafadam, 2025-04-09:
Hi, CameraInfo messages are converted from our internal calibration here https://github.com/luxonis/depthai-ros/blob/9c1682aed12ca3f986d3d4a31d8ccf7b7295b7c3/depthai_bridge/src/ImageConverter.cpp#L428
They should match correctly for all resolutions/image sizes, if that is not the case please let us know.
You can also check the calibration on the device with camera.i_calibration_dump: true` parameter, or via Python/C++ examples
The calibration files you mentioned are an example calibration files obtained by using ROS calibration packages. In general, you should use our internal calibration since it is more accurate but sometimes (although it's pretty rare now) libraries require `plumb_bob` distortion model which is a bit different than our `rational_polynomial` in that case you can recalibrate in ROS and override the CameraInfo via those files.

khuechuong, 2025-04-09:
@Serafadam Here's the raw and rect image:

[image]

The rect is definitely better than raw, but I think the edge wall line close to the door on the right looks like it's not 100% straight.

khuechuong, 2025-04-09:
Do you guys know what the HFOV and VFOV after rect?