luxonis/depthai-ros #710: [BUG] No pointcloud (/oak/points) being published when launching rgbd-pcl.launch.py

lts0429, 2025-05-06:
**Check if issue already exists**

I found this issue but it was closed due to inactivity https://github.com/luxonis/depthai-ros/issues/303

**Describe the bug**
I ran the following in my terminal
```
ros2 launch depthai_ros_driver rgbd_pcl.launch.py use_rviz:=true
```
No pointcloud is displayed in rviz. When I ran ros2 topic echo /oak/points, I didn't receive anything

Here is the output
```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [70239]
[INFO] [component_container-2]: process started with pid [70241]
[component_container-2] [INFO] [1746519182.331325995] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-2] [INFO] [1746519182.333544090] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-2] [INFO] [1746519182.333564317] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-2] [INFO] [1746519182.338695388] [oak_state_publisher]: got segment oak
[component_container-2] [INFO] [1746519182.338714419] [oak_state_publisher]: got segment oak-d-base-frame
[component_container-2] [INFO] [1746519182.338718359] [oak_state_publisher]: got segment oak_imu_frame
[component_container-2] [INFO] [1746519182.338721479] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-2] [INFO] [1746519182.338724898] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-2] [INFO] [1746519182.338727836] [oak_state_publisher]: got segment oak_model_origin
[component_container-2] [INFO] [1746519182.338730801] [oak_state_publisher]: got segment oak_rgb_camera_frame
[component_container-2] [INFO] [1746519182.338733770] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[component_container-2] [INFO] [1746519182.338736654] [oak_state_publisher]: got segment oak_right_camera_frame
[component_container-2] [INFO] [1746519182.338739604] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-2] [INFO] [1746519182.339925632] [oak_container]: Load Library: /opt/ros/humble/lib/libdepthai_ros_driver.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container '/oak_container'
[component_container-2] [INFO] [1746519182.383766239] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1746519182.383810908] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1746519182.386066953] [oak_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-2] [INFO] [1746519182.388231338] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1746519182.388248690] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-2] [INFO] [1746519182.388252438] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-2] [INFO] [1746519182.388255632] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-2] [INFO] [1746519182.388260168] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-2] [INFO] [1746519182.388263467] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-2] [INFO] [1746519182.388267012] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziRadialNode>
[component_container-2] [INFO] [1746519182.388270150] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[component_container-2] [INFO] [1746519182.388273515] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzrgb_node' in container '/oak_container'
[rviz2-1] [INFO] [1746519182.805808545] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1746519182.805919971] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1746519182.992170260] [rviz2]: Stereo is NOT SUPPORTED
[component_container-2] [WARN] [1746519183.042567876] [point_cloud_xyzrgb_node]: New subscription discovered on topic '/oak/points', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container-2] [INFO] [1746519183.387040471] [oak]: Starting camera.
[component_container-2] [INFO] [1746519183.389350321] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [INFO] [1746519186.106233816] [oak]: Camera with MXID: 14442C105121D2D200 and Name: 3.2 connected!
[component_container-2] [INFO] [1746519186.106902500] [oak]: USB SPEED: SUPER
[component_container-2] [INFO] [1746519186.152742184] [oak]: Device type: OAK-D-PRO-W
[component_container-2] [INFO] [1746519186.154757729] [oak]: Pipeline type: RGBD
[component_container-2] [INFO] [1746519186.811979568] [oak]: Finished setting up pipeline.
[component_container-2] [INFO] [1746519187.164445128] [oak]: Camera ready!

```
**Minimal Reproducible Example**
- Installed depthai ros driver using binaries (sudo apt-get install ros-humble-depthai-ros)
- Ran the example launch file (ros2 launch depthai_ros_driver rgbd_pcl.launch.py use_rviz:=true)

**Attach system log**
os: Ubuntu 22.04
ros-distro: ros2 humble
`depthai-ros` installed from apt

fabrisdnl, 2025-05-29:
I'm encountering the same policy problem using a OAK D SR POE, I'm subscribed to the **/oak/points** topic, using a sensor_msgs.msg PointCloud2 to store data.
In my code i have written:

from rclpy.qos import QoSProfile, ReliabilityPolicy, HistoryPolicy

pc_qos = QoSProfile(
    reliability=ReliabilityPolicy.BEST_EFFORT,
    history=HistoryPolicy.KEEP_LAST,
    depth=10
)

self.create_subscription(
    PointCloud2,
    '/oak/points',
    self.pointcloud_callback,
    qos_profile=pc_qos
)

The error is:
[component_container-5] [INFO] [1748512479.248356506] [oak]: Camera ready!
[component_container-5] [WARN] [1748512479.248872537] [point_cloud_xyzright_node]: New subscription discovered on topic '/oak/points', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: **RELIABILITY_QOS_POLICY**.
Did anyone solve this problem or have any suggestion?

gjsinicr, 2025-05-29:
Same issue, tried to change the QOS policy but it’s read only. This worked before with Jazzy, I have an older install, I will revive it and see if it still works, however I did upgrade the firmware on the POE camera to the latest .28.

MattiaDeiRossi, 2025-05-29:
Hi all, 

I have the same issue with the same model of @fabrisdnl (OAK D SR POE).
Do you have any update about it?

Thanks

Hurisa, 2025-07-02:
Hi All,

I'm also having this issue with the oak-d lite camera.
However, if I launch `ros2 launch depthai_ros_driver pointcloud.launch.py` I can see the pointcloud on foxglove. Just not overlaid with the rgb.

Any help is much appreciated.
Thanks

Serafadam, 2025-07-03:
Hi, `rgbd_pcl.launch.py` and most likely `pointcloud.launch.py` will be deprecated in new releases, to get RGB Pointclouds you can now use `camera.launch.py` with `pointcloud.enable:=true` argument. Also `pointcloud.launch.py` does not provide RGB pointclouds, just ones based on intensity. Regarding reliability policy messages, these most likely come from subscribing to the messages in Rviz and can be ignore after setting correct topic policy in Rviz itself.

closed 2025-07-04 by lts0429

longuyen3127, 2025-07-12:
> Hi, `rgbd_pcl.launch.py` and most likely `pointcloud.launch.py` will be deprecated in new releases, to get RGB Pointclouds you can now use `camera.launch.py` with `pointcloud.enable:=true` argument. [...]

I'm getting this same issue with camera.launch.py as well. Running it with rviz off but nothing is being published to oak/points.